eclipse-mraa/mraa #660: Bug in the python binding for I2C read

jeremypatrickdahan, 2017-03-05:
I found a bug in the generated python binding : the signature is not correct  

(from https://iotdk.intel.com/docs/master/mraa/python/_modules/mraa.html#I2c.read )
```python
    def read(self, data):
        """
        read(I2c self, uint8_t * data) -> int

        Parameters
        ----------
        data: uint8_t *

        """
        return _mraa.I2c_read(self, data)
```
 
should be 

```python
def read(self, length):
    """
    read(I2c self, int length) -> int

    Parameters
    ----------
    length: int
  
    """
    return _mraa.I2c_read(self, length)
```

Otherwise the function will raise an error saying either the number of arguments or their types is not correct.

To reproduce the bug, just call the function with an arbitrary array :

```
data = bytearray([0]*10)
magSensor.read(data)
```

I am not familiar with SWIG so I can't locate the error in the original code

alext-mkrs, 2017-03-05:
This seems to be the same as #645 (though a different function).

jeremypatrickdahan, 2017-03-05:
Well, #645 is a matter of documentation but here the function cannot be called at all through the python API.

alext-mkrs, 2017-03-05:
Well, it's actually identical if you look at the example there. Same problem - instead of "int length" docs got "uint8 * data".

jeremypatrickdahan, 2017-03-05:
Sorry, I didn't read carefully !

alext-mkrs, 2017-03-07:
Ok, let me close this one as duplicate then and #645 will stay as a master.

closed 2017-03-07 by alext-mkrs