eclipse-mraa/mraa #663: [FR]switch from node-gyp to node-pre-gyp

farfromrefug, 2017-03-07:
That would be awesome if you could switch from node-gyp to [node-pre-gyp](https://github.com/mapbox/node-pre-gyp).
that way we could have pre build release and make ```npm install``` so much faster!
If you need an example [node-usb](https://github.com/tessel/node-usb) uses it

Thanks

arfoll, 2017-03-07:
That's a really good idea, I did not know that existed, any ideas how we can quickly do that? just node-pre-gyp publish?

farfromrefug, 2017-03-07:
you have to change the package.json and the binding.gyp.
As an example:
[this](https://github.com/tessel/node-usb/blob/master/package.json#L38) is to set the host were to look for pre built
[this](https://github.com/tessel/node-usb/blob/01f59d6472540ff61fa01c0ac769a8334322a040/binding.gyp#L8) is for the actual node-pre-gyp build

PS: does a good name to "module_name" like simply "mraa". This is used by node-pre-gyp to allow override of host url.

arfoll, 2017-03-07:
Ok, thanks for the info I'll have to dig a little bit.

Ping @intel-iot-devkit/upm-core-dev this looks pretty interesting!