eclipse-mraa/mraa #669: mraa_pwm_close( ) API not working

vinugk, 2017-03-10:
The mraa library api mraa_pwm_close( ) seems to not working. It returns success, however the file descriptor is not freed. I have a requirement of init and close the pwm a number of times. With the close function not working i find that Intel Edison ends up in memory leakage. Please help

alext-mkrs, 2017-03-10:
Which descriptor do you mean? I've just scanned through the code and I don't see anything like that (with a disclaimer that I did that on my phone, which is not the best tool for the job, but all I have right now).

vinugk, 2017-03-13:
mraa_pwm_init( ) creates a file handler whenever called. This file handler normally should have been closed by mraa_pwm_close( ). However this is not happening, which means every time mraa_pwm_init() is called a new file handler is created and results in memory leakage

alext-mkrs, 2017-03-13:
Ok, so what you actually mean is mraa_pwm_setup_duty_fp(), which opens the duty sysfs file and indeed that one is never closed. That's a bug, marking as such.

vinugk, 2017-03-13:
Thanks @alext-mkrs . Is this a known issue? Are there any possible workarounds or fixes available?

alext-mkrs, 2017-03-13:
Nope, it's not a known one, your report is the first one. There's no workaround I can think of, but just fixing it should be fairly trivial. If no one beats me to it, I plan to prepare a fix tomorrow.

commit 21e0ff7c (2017-03-16): pwm.c: properly close duty file when disabling PWM

Fixes #669.
closed 2017-03-16 by arfoll