eclipse-mraa/mraa #670: Command 'mraa-gpio set' fails randomly on Joule

skuusela, 2017-03-13:
Using `mraa gpio set` command in terminal to set a GPIO pin randomly fails (about 1% of the time) causing an error and not changing the GPIO pin state. Tested it using Yocto based [Intel Refkit](https://github.com/intel/intel-iot-refkit) image and Joule BIOS version 161. Also tried different GPIO pins with the same result. There were no errors when using Minnowboard Turbot and the same image. Error message with Joule:
```
*** Error in `mraa-gpio': free(): invalid pointer: 0x00000000006193f5 ***
======= Backtrace: =========
/lib/libc.so.6(+0x7039b)[0x7f34b9aa739b]
/lib/libc.so.6(+0x76826)[0x7f34b9aad826]
/lib/libc.so.6(+0x7701e)[0x7f34b9aae01e]
/usr/lib/libmraa.so.1(mraa_find_i2c_bus_pci+0x22f)[0x7f34b9ddc77f]
/usr/lib/libmraa.so.1(mraa_gt_tuchuck_board+0xd0)[0x7f34b9df0d60]
/usr/lib/libmraa.so.1(mraa_x86_platform+0x3c3)[0x7f34b9de8753]
/usr/lib/libmraa.so.1(imraa_init+0x1ef)[0x7f34b9ddb8cf]
/lib/ld-linux-x86-64.so.2(+0xf1ba)[0x7f34ba00c1ba]
/lib/ld-linux-x86-64.so.2(+0xf2cb)[0x7f34ba00c2cb]
/lib/ld-linux-x86-64.so.2(+0xc7a)[0x7f34b9ffdc7a]
======= Memory map: ========
00400000-00403000 r-xp 00000000 b3:03 4655                               /usr/bin/mraa-gpio
00602000-00603000 r--p 00002000 b3:03 4655                               /usr/bin/mraa-gpio
00603000-00604000 rw-p 00003000 b3:03 4655                               /usr/bin/mraa-gpio
00604000-00625000 rw-p 00000000 00:00 0                                  [heap]
7f34b4000000-7f34b4021000 rw-p 00000000 00:00 0 
7f34b4021000-7f34b8000000 ---p 00000000 00:00 0 
7f34b8fd7000-7f34b8fed000 r-xp 00000000 b3:03 4308                       /lib/libgcc_s.so.1
7f34b8fed000-7f34b91ec000 ---p 00016000 b3:03 4308                       /lib/libgcc_s.so.1
7f34b91ec000-7f34b91ed000 r--p 00015000 b3:03 4308                       /lib/libgcc_s.so.1
7f34b91ed000-7f34b91ee000 rw-p 00016000 b3:03 4308                       /lib/libgcc_s.so.1
7f34b91ee000-7f34b9202000 r-xp 00000000 b3:03 4299                       /lib/libnsl-2.25.so
7f34b9202000-7f34b9402000 ---p 00014000 b3:03 4299                       /lib/libnsl-2.25.so
7f34b9402000-7f34b9403000 r--p 00014000 b3:03 4299                       /lib/libnsl-2.25.so
7f34b9403000-7f34b9404000 rw-p 00015000 b3:03 4299                       /lib/libnsl-2.25.so
7f34b9404000-7f34b9406000 rw-p 00000000 00:00 0 
7f34b9406000-7f34b940d000 r-xp 00000000 b3:03 887                        /lib/libnss_compat-2.25.so
7f34b940d000-7f34b960c000 ---p 00007000 b3:03 887                        /lib/libnss_compat-2.25.so
7f34b960c000-7f34b960d000 r--p 00006000 b3:03 887                        /lib/libnss_compat-2.25.so
7f34b960d000-7f34b960e000 rw-p 00007000 b3:03 887                        /lib/libnss_compat-2.25.so
7f34b960e000-7f34b9618000 r-xp 00000000 b3:03 10396                      /usr/lib/libjson-c.so.2.0.1
7f34b9618000-7f34b9817000 ---p 0000a000 b3:03 10396                      /usr/lib/libjson-c.so.2.0.1
7f34b9817000-7f34b9818000 r--p 00009000 b3:03 10396                      /usr/lib/libjson-c.so.2.0.1
7f34b9818000-7f34b9819000 rw-p 0000a000 b3:03 10396                      /usr/lib/libjson-c.so.2.0.1
7f34b9819000-7f34b9832000 r-xp 00000000 b3:03 4295                       /lib/libpthread-2.25.so
7f34b9832000-7f34b9a31000 ---p 00019000 b3:03 4295                       /lib/libpthread-2.25.so
7f34b9a31000-7f34b9a32000 r--p 00018000 b3:03 4295                       /lib/libpthread-2.25.so
7f34b9a32000-7f34b9a33000 rw-p 00019000 b3:03 4295                       /lib/libpthread-2.25.so
7f34b9a33000-7f34b9a37000 rw-p 00000000 00:00 0 
7f34b9a37000-7f34b9bcb000 r-xp 00000000 b3:03 2444                       /lib/libc-2.25.so
7f34b9bcb000-7f34b9dca000 ---p 00194000 b3:03 2444                       /lib/libc-2.25.so
7f34b9dca000-7f34b9dce000 r--p 00193000 b3:03 2444                       /lib/libc-2.25.so
7f34b9dce000-7f34b9dd0000 rw-p 00197000 b3:03 2444                       /lib/libc-2.25.so
7f34b9dd0000-7f34b9dd4000 rw-p 00000000 00:00 0 
7f34b9dd4000-7f34b9dfb000 r-xp 00000000 b3:03 6198                       /usr/lib/libmraa.so.1.5.1
7f34b9dfb000-7f34b9ffb000 ---p 00027000 b3:03 6198                       /usr/lib/libmraa.so.1.5.1
7f34b9ffb000-7f34b9ffc000 r--p 00027000 b3:03 6198                       /usr/lib/libmraa.so.1.5.1
7f34b9ffc000-7f34b9ffd000 rw-p 00028000 b3:03 6198                       /usr/lib/libmraa.so.1.5.1
7f34b9ffd000-7f34ba020000 r-xp 00000000 b3:03 4292                       /lib/ld-2.25.so
7f34ba211000-7f34ba213000 rw-p 00000000 00:00 0 
7f34ba21c000-7f34ba21f000 rw-p 00000000 00:00 0 
7f34ba21f000-7f34ba220000 r--p 00022000 b3:03 4292                       /lib/ld-2.25.so
7f34ba220000-7f34ba221000 rw-p 00023000 b3:03 4292                       /lib/ld-2.25.so
7f34ba221000-7f34ba222000 rw-p 00000000 00:00 0 
7fffe40f8000-7fffe4119000 rw-p 00000000 00:00 0                          [stack]
7fffe41ac000-7fffe41ae000 r--p 00000000 00:00 0                          [vvar]
7fffe41ae000-7fffe41b0000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
Aborted (core dumped)

```

arfoll, 2017-03-13:
So that's a bit odd, especially if it doesn't happen all the time, it looks like it fails in the init phases whilst trying to probe i2c busses. Is it possible to get a proper gdb backtrace to see what line it fails at? Obviously the fail is somewhere here: https://github.com/intel-iot-devkit/mraa/blob/master/src/mraa.c#L932 but hard to tell exactly where. It's also very odd that it's semi random...

arfoll, 2017-03-13:
@Propanu you seen this at all with your recent Joule playing?

alext-mkrs, 2017-03-13:
I've used it a bit recently, with BIOS 193 but working on UART stuff, haven't noticed any init problems, FWIW.

skuusela, 2017-03-13:
@arfoll To my knowledge the 161 is the newest public BIOS version but I tried with 193 and also with newest 1A0 BIOS version and the problem seemed to persist. I also tried mraa with Ubuntu 16.04 and couldn't get the error with 1000 tries. So the problem seems to be related to Refkit.

arfoll, 2017-03-13:
@skuusela likely there is an error condition not being handled somewhere in sysfs, can you find out exactly on which lines it blows up with gdb? I can't imagine the fix will be too difficult but it's odd it's not reproducable all the time...
@arunlee77 why are non public bioses being tested and patches submitted against those?

skuusela, 2017-03-13:
@arfoll Yes, I'll try with gdb. Just need to build an image with it. Maybe testing with non public BIOS version makes sense as when they get released publicly everything already works with them?

arunlee77, 2017-03-13:
@arfoll  For the GPIO patch i submitted, the mapping is valid for older BIOSes and Ostro images.

arfoll, 2017-03-13:
@arunlee77 ok happy to hear that.

skuusela, 2017-03-14:
@arfoll [#673](https://github.com/intel-iot-devkit/mraa/pull/673) solves the random crashing issue with Joule. Also tagging @ipuustin as he helped me with the issue.

closed 2017-03-17 by skuusela