eclipse-mraa/mraa #681: Feature Request: detect platform in cmake compilation

nicolalunghiemutex, 2017-03-23:
I've found that I can actually detect the board that I'm in doing the following in cmake

```
file(STRINGS` "/sys/class/dmi/id/board_name"    mraa_BOARD_NAME)
file(STRINGS "/sys/class/dmi/id/board_version" mraa_BOARD_VERSION)
message(STATUS "INFO - Board Name String: ${mraa_BOARD_NAME}")
message(STATUS "INFO - Board Version String: ${mraa_BOARD_VERSION}")

if (mraa_BOARD_NAME STREQUAL "UP-CHT01" AND mraa_BOARD_VERSION STREQUAL "V1.0")
    message(STATUS "INFO - Detected UP platform")
    set(MRAA_PLATFORM_UP ON)
endif() 
```
and in the test directory in the CMakeLists.txt I can say
```
if (X86PLAT)
    if(MRAA_PLATFORM_UP)
        add_subdirectory(up)
    endif()
endif()
```
or also in the x86 directory and compile only the sources relative to the platform I'm on

I don't know if this modification could impact other boards also???

nicolalunghiemutex, 2017-03-23:
Also is better to put it in the main CMakeLists.txt or in the one of the test directory? is useful only for selecting which test to run or also could be useful for example for some conditional compilation? eg compile only the part related to my board

 list(APPEND X86_SRCS "")
if(MRAA_PLATFORM_EDISON)
 list(APPEND X86_SRCS edison.c)
endif()

arfoll, 2017-03-23:
That is actually a supercool idea, and it would be great to have a flag -DDETECTPLAT=ON! It's a good complement to  our current -DMRAAPLATFORMFORCE flag. However if you do this, it needs to be done on most/all platforms not just one. I feel that we still need a force flag as cross compilation is a common use case for these boards.

For tests however this is a great idea, even if only a few/one platforms are supported.

Generally platforms (bar the arch diffs which make no sense to include), are all included in a mraa build, this is so you can copy the binaries of libmraa.so from one computer to the next, unless you're on a very targeted yocto/buildroot build of Linux where you know exactly where you are, it doesn't make much sense to split it and I don't really want to see x86 boards outside of the x86/ folder if it's not necessary.

nicolalunghiemutex, 2017-03-24:
https://github.com/intel-iot-devkit/mraa/pull/683

closed 2017-03-24 by nicolalunghiemutex