eclipse-mraa/mraa #703: Porting mraa for the CHIP

teuteuguy, 2017-04-10:
Could we add the NextThing Chip to the list of supported devices?

Happy to help test, run, modify files to test it out, if someone tells me what to do.

teuteuguy, 2017-04-12:
If someone can guide on how to work prepare build files for new platform like the CHIP, it's something I can try to work and test. But don't know where to start or how to do it, test it.

alext-mkrs, 2017-04-13:
You genetally need a platform/board file as the core - see e.g. src/x86 or src/arm. You can also check the commit history for commits adding various platforms.

TechNexion, 2017-05-17:
In general, what is the policy on accepting patches for (new) boards? We have a few we might want to port - but before doing so it might be good to get a hint if this is anything anyone is interested in.

arfoll, 2017-05-17:
If you look at our contributing.md guide it starts by saying we'll be happy to add any linux platform to mraa :) As such we are very keen to support as many platforms as possible so please add away :)

teuteuguy, 2017-07-14:
Is there documentation available on how to go about adding a board ?
I'd try, but have no idea what needs to be done

arfoll, 2017-07-14:
If you look in docs/internal.md and related there are some details. I'd
probably recommend taking a random board file in src/arm or src/x86, and
starting very simple. Have a look at a commit that adds a board file. When
I get back to a computer I'll have a look

arfoll, 2017-07-14:
Have a look at this commit - it has pretty much everything you need to add a platform. If you are unsure of any details - feel free to ask! https://github.com/intel-iot-devkit/mraa/commit/4472ff162989273ed86a97bbd1288494e09a5782