eclipse-mraa/mraa #705: python api: impossible to get uart and spi count

nicolalunghiemutex, 2017-04-11:
Hi
how I can from python get the number of uart and spi busses?
and equivalent of 
mraa.getI2cBusCount()

Regards
Nicola

alext-mkrs, 2017-04-11:
There's no such API for SPI and UARTs, patches are welcome :-)  There are such fields in the plat structure, but nowrappers, like for I2C. Why would you need that?

arfoll, 2017-04-11:
Well ideally we'd have a generic function where you ask int
mraa_get_io_count(mraa_pinmodes_t). That could be quite useful.

nicolalunghiemutex, 2017-04-11:
I'm writing something similar to this

# get gpio pins
MRAA_GPIO_PINS = [(mraa.getPinName(i), i) for i in range(mraa.getPinCount()) if mraa.pinModeTest(i, mraa.PIN_GPIO)]

I need to enumerate the UART and spi defined on the board and print it / permit to select it