eclipse-mraa/mraa #712: Error while Run code for Grove LCD in intel edison

salmanfarisvp, 2017-04-26:
i got an error while uploading code for Grove LCD in  intel edison , i updated all 

**code** :-
```
// change this to false to use the hand rolled version
var useUpmVersion = true;

// we want mraa to be at least version 0.6.1
var mraa = require('mraa');
var version = mraa.getVersion();

if (version >= 'v0.6.1') {
    console.log('mraa version (' + version + ') ok');
}
else {
    console.log('meaa version(' + version + ') is old - this code may not work');
}

if (useUpmVersion) {
    useUpm();
}
else {
    useLcd();
}

/**
 * Rotate through a color pallette and display the
 * color of the background as text
 */
function rotateColors(display) {
    var red = 0;
    var green = 0;
    var blue = 0;
    display.setColor(red, green, blue);
    setInterval(function() {
        blue += 64;
        if (blue > 255) {
            blue = 0;
            green += 64;
            if (green > 255) {
                green = 0;
                red += 64;
                if (red > 255) {
                    red = 0;
                }
            }
        }
        display.setColor(red, green, blue);
        display.setCursor(0,0);
        display.write('red=' + red + ' grn=' + green + '  ');
        display.setCursor(1,0);
        display.write('blue=' + blue + '   ');  // extra padding clears out previous text
    }, 1000);
}

/**
 * Use the upm library to drive the two line display
 *
 * Note that this does not use the "lcd.js" code at all
 */
function useUpm() {
    var lcd = require('jsupm_i2clcd');
    var display = new lcd.Jhd1313m1(0, 0x3E, 0x62);
    display.setCursor(1, 1);
    display.write('hi there');
    display.setCursor(0,0);
    display.write('more text');
    rotateColors(display);
}

/**
 * Use the hand rolled lcd.js code to do the
 * same thing as the previous code without the
 * upm library
 */
function useLcd() {
    var lcd = require('./lcd');
    var display = new lcd.LCD(0);

    display.setColor(0, 60, 255);
    display.setCursor(1, 1);
    display.write('hi there');
    display.setCursor(0,0);  
    display.write('more text');
    display.waitForQuiescent()
    .then(function() {
        rotateColors(display);
    })
    .fail(function(err) {
        console.log(err);
        display.clearError();
        rotateColors(display);
    });
}
`

**Error**
```

`TypeError: lcd.Jhd1313m1 is not a function
    at useUpm (/home/root/.node_app_slot/main.js:75:19)
    at Object.<anonymous> (/home/root/.node_app_slot/main.js:32:5)
    at Module._compile (module.js:409:26)
    at Object.Module._extensions..js (module.js:416:10)
    at Module.load (module.js:343:32)
    at Function.Module._load (module.js:300:12)
    at Function.Module.runMain (module.js:441:10)
    at startup (node.js:139:18)
    at node.js:968:3`

alext-mkrs, 2017-04-27:
This is rather for upm repo, not mraa. But anyway, looks like your "require" doesn't work. Are you sure upm is installed properly? Please check in the node interactive mode, would that "require" return proper object?

jontrulson, 2017-04-28:
Please note, in the latest UPM (1.2.0), jhd1313m1 has been moved into it's own separate library 'jhd1313m1', rather than the old library 'i2clcd'...

`var lcd = require('jsupm_jhd1313m1');`

...is likely to work better.

salmanfarisvp, 2017-05-03:
@jontrulson Thanks ,now working fine

closed 2017-05-03 by salmanfarisvp