eclipse-mraa/mraa #713: using new chardev access for gpios in new kernel

Mani-Sadhasivam, 2017-04-27:
In kernel 4.7, new chardev access has been added which allows the userspace to access the gpios in an efficient way compared to sysfs interface. 

There is already a library available (https://github.com/brgl/libgpiod). So, I'd like to add support for this new way of accessing gpio in libmraa for kernels > 4.7

arfoll, 2017-04-27:
This makes a lot of sense, please go ahead :)

Mani-Sadhasivam, 2017-05-04:
How should we distinguish the kernel version for implementing switch (sysfs/chardev)? 
It will become very hard if the library is cross compiled. So, uname won't work!
env variable maybe?

bjbeare, 2017-05-04:
Can presence of the node act as the switch?

Mani-Sadhasivam, 2017-05-04:
It can! Just want to know whether to stick with static detection or dynamic :) Will stick with dynamic detection

arfoll, 2017-05-04:
Dynamic detection is the best and is in keeping with how mraa generally works. My recommendation would be to check for the kernel version first and then if kernel looks like it could have the chardev node then try access it.

Then we can add some cmake options to disable one or the other modes.

FYI please add this as a full implementation inside src/gpio.c as src/gpio_chardev.c and src/gpio_sysfs.c rather than just making src/gpio.c too big and make src/gpio.c just the code that is the same for both/malloc of the _gpio context etc...

Propanu, 2017-05-04:
@arfoll what's your take on adding char* device paths to all the MRAA I/O structs (incl. GPIO). It seems like that's one of the things we want to easily customize, but I haven't considered the performance/size hit.

RobertBerger, 2017-12-11:
Just a stupid question:"What don't we build on top of libgpiod?" [1]

"Since linux 4.8 the GPIO sysfs interface is deprecated. User space should use
the character device instead. This library encapsulates the ioctl calls and
data structures behind a straightforward API."

I somehow get the feeling that things are re-invented here.

[1] https://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git/

bjbeare, 2017-12-11:
The base kernel support requirement for Android Things is 4.4.

RobertBerger, 2017-12-11:
@bjbeare Now I am even more confused. In this thread we are __NOT__ talking about 4.4 and also __NOT__ about whatever Android Things uses. It's about the new chardev access for gpios.

bjbeare, 2017-12-11:
Perhaps I should have phrased my comment as a question...

Is there a version of libgpiod that works well with a 4.4 kernel that has no backports to support it?

Mani-Sadhasivam, 2017-12-12:
@RobertBerger So the issue is Licensing. The day when I started to work on this support, `libgpiod` was licensed under **GPL v3** and now it has been changed to **LGPL v2.1+**. So, @arfoll didn't favour dependency with the incompatible license and that's why this whole reinvention came up.

One interesting thing about `libgpiod` is that it has grown from a normal userspace tool to a well structured library now!

arfoll, 2017-12-12:
@Mani-Sadhasivam is right, it was mostly driven out of the license. GPLv3 was a not appropriate for us, LGPLv2.1 is ok but still not as open as we'd usually like.

I don't really see it as re-invention, chardev is a public kernel API and frankly it doesn't take that much code to use it. Mraa is meant as a low level library and so tries to limit it's dependencies.

Mani-Sadhasivam, 2018-02-23:
@arfoll Any update on this? It might be better to switch to new Chardev interface quicker. Lemme know if you need any kind of help from me!

Mani-Sadhasivam, 2018-02-26:
@arfoll Just realised that most of the 96Boards platforms got upstream support and they will be running close to mainline. Which means, sysfs way of accessing GPIO's won't work anymore. This is due to the fact that the GPIO base will be set dynamically by the gpiolib core and we can no more map GPIOs using global number space and the only resort is to use Chardev interface.

Please consider Chardev merge as a priority work if possible...