eclipse-mraa/mraa #721: writeByte throws Unspecified Error

matthewmax09, 2017-05-05:
Hello,

I am writing an I2c Driver for my Edison to read flow data from the Px4Flow sensor. The [Px4Flow I2c](https://pixhawk.org/modules/px4flow#i2c) page says that to read the I2C frame data I need to send 0x0 to the sensor.
> send 0x0 to PX4FLOW module and receive back 22 Bytes data

Below is the snippet of the code I am using to read from the sensor. But at the point where I write 0x0 to the bus, I receive an error: **MRAA: Unspecified Error.** I am hoping someone here could advise if this error is due to the wrong use of "writeByte"?  

As a reference guide for my code, I am using this [MRAA C++ example](https://github.com/intel-iot-devkit/mraa/blob/master/examples/c%2B%2B/I2c-compass.cpp).

```
mraa::I2c* i2c    

i2c = new mraa::I2c(1);

i2c->address(Px4Flow_I2C_ADDR);

while (ros::ok()) {
	
	i2c->address(Px4Flow_I2C_ADDR);
	
	i2c->writeByte(Px4Flow_I2C_Readout);
	
	mraa::printError(i2c->writeByte(Px4Flow_I2C_Readout));

	i2c->address(Px4Flow_I2C_ADDR);
	
	i2c->read(rx_tx_buf, DATA_REG_SIZE);
	
}
```

alext-mkrs, 2017-05-05:
Two things to check out, I'd say:

1. You're effectively calling `writeByte(Px4Flow_I2C_Readout)` twice in your code, is that intentional?
1. check the `journalctl -xn 40 --no-pager` output after getting the error - it may contain additional clues either from the library itself or kernel/drivers.

arfoll, 2017-05-08:
> @alext-mkrs <https://github.com/alext-mkrs> Yes. you're right, it is not
> intentional. I have removed the line i2c->writeByte(Px4Flow_I2C_Readout);.
>
> sadly my distro , Ubilinux, does not have journalctl installed. Do you
> know other alternatives to view the output?

Mraa logs to syslog so check in /var/log/messages or a similar file

matthewmax09, 2017-05-09:
@arfoll Thanks for that. I found out through the logs that it was a I/O access error and was able to quickly rectify it. Thank you for your help!

alext-mkrs, 2017-05-09:
Good to read that and thanks for reporting back!

closed 2017-05-09 by alext-mkrs