eclipse-mraa/mraa #732: Ownership of pointers in mraa_board_t

TechNexion, 2017-05-17:
It seems like there is no clear definition on who owns the different pointers (usually `char*`) inside `mraa_board_t`.
Frequently, they are assigned a constant value (i.e. `b->platform_name = "MonkeyBoard";`) and then freed with `free(b->platform_name)`. Not nice, and might cause segfaults or C-library stack dumps on `mraa_deinit()`.

Same applies for other character pointers, like `device_path` in `mraa_uart_dev_t`.

As a side note, since I don't know what is the proper place to discuss overall items with this library:

Has anyone run a static code analyzer on this library? Or a runtime memory integrity tester (like valgrind)? Considering how many bugs can be spotted just by attempting to use the library and by glancing at the code, I think using some memory error detection tools might be a useful (and maybe scary) exercise to bring the code closer to production level.

arfoll, 2017-05-17:
Yes, this is not nice. mraa_deinit() in general is not great and works realtively badly. It's rare that mraa is unloaded like this so it's never been at the forefront of development, but it should be improved.

Originally the names where all constants, this has changed a little over time with newer library additions and sadly some bugs have been introduced :/

Actually we've run a bunch of static analysis tools and generally they don't complain too much, @alext-mkrs is currently setting up a public instance of sonarcube and fixing a bunch of them.

Generally it is the responsibility of the platform files. Generally I think the easiest thing would be to have a deinit() per platform be called as part of an advance function hook so that a platform always deinit() correctly.

TechNexion, 2017-05-17:
Letting the platforms free what they allocate sounds like a much better way to go to me. 
Maybe introduce platform destructor functions in x86.c and arm.c, which in turn can mux the destructor call to the actual platform file. In case the platform has no destructor, either a default deinit can be used (like now) or just exit without deinit.

arfoll, 2017-05-17:
There's actually no need to mux this, we can just use the advance function and add a mraa_deinit_replace. Although having had a deeper look into what you are saying I don't actually believe we even have an issue (although I'm sure there's an edge case somewhere, as stated above mraa_deinit() is not tested that thoroughly and likely has some edge cases in some of our many build/runtime configurations...)

1. No platforms allocates plat->platform_name, they all use constants. The exception is the json_platform which does allocate it, but this is treated in mraa_deinit as a special case with mraa.c:263
2. platform_name is freed (not mraa_board_t.platform_name) but this is always allocated see mraa.c:166 or mraa.c:204 as it's a composite name of mraa_board_t.platform_name and a possible subplatform or something from a json file.
3. The freeing of device_path only occurs when platform_type is MRAA_JSON_PLATFORM.  But this is correct as they are allocated see jsonplatform.c:525

TechNexion, 2017-05-18:
You seem to be right. Did not expect a global variable with exactly the same name as the `board_t` member variable (your point 2) and mixed them up - being new to the code base and all.
Maybe the deinit part of the code is ok, I've not seen an actual issue. Just found this when filling out the `mraa_board_t` struct for my board (and tried to figure out who owns the pointers there).

This topic can be closed from my point of view.

arfoll, 2017-05-18:
yeah I probably should have called it combined_platform_name or something, probably not the smartest bit of naming i've done :)

closed 2017-05-18 by arfoll