eclipse-mraa/mraa #734: UART setting change hang?

TechNexion, 2017-05-18:
Worried I've struck a quite severe bug in the UART layer.

Changing baudrate or flow control can hang the process quite hard (uninterruptible/disk sleep). Any subsequent accesses (from different processes obviously) to the same UART device also hangs (i.e. reboot needed). 
The same setting changes using standard linux command line tools (like stty) appears to have no such problems. This leads me to suspect the MRAA library or the way it operates the UART.
**Affected systems:** reproducable under Ubuntu 16.04 on both 32-bit ARM and a x86-64 desktop, running kernels 4.1 and 4.4 respectively. UART can be native one or external USB serials.
**Steps to reproduce:** change settings like baudrate/ctsrts many times randomly. The UART can with benefit be a USB serial (used in raw mode). This way, detaching the USB cable is a way to recover avoiding rebooting.

Causes are unknown. The process hangs in an `ioctl()` originating from `tcsetattr` call from uart.c. Changing the `tcsetattr` update modes from `TCSAFLUSH` to `TCSANOW` lessens the frequency of the problem and after that change there is a chance the process can wake up again (after waiting 30 seconds to several minutes).
On the ARM board there were kernel errors about RCU stalls when the hang occured.

Attaching my shell script to stress a UART using linux stty. Can share my C-code using MRAA, but maybe it is better to be tested independently. It is possible that is where the true problem lies...

-----------------------------
```
#!/bin/sh

uart=$1
if [ X$uart = X ]
then
        uart=/dev/ttyUSB0
fi

if [ ! -c $uart ]
then
        echo No UART $uart exist
        exit 1
fi

its=0
while [ $its -lt 100 ]
do
        for flow in crtscts -crtscts
        do
                for baud in 9600 115200 38400 57600 2400 19200
                do
                        stty -F ${uart} ${baud} ${flow}
                        sleep 0.001
                done
        done
        its=`expr $its + 1`
done

```

alext-mkrs, 2017-05-18:
mraa-based code would be useful anyway to make sure the tests are the same and we are not checking different things.

Aside from anything else - I wonder what's the use case for changing UART parameters every millisecond?

arfoll, 2017-05-18:
Tbh USB serial devices, especially the cheaper pl203 based devices are pretty crash prone in my experience. FTDI based devices are usually more robust. But it's very odd.

In any case as Linus would say - we shouldn't manage to crash the kernel driver from userspace so there is a bug in the kernel (reguardless of whether we are doing things wrong or not!).

TechNexion, 2017-05-19:
@alext-mkrs : the use case for changing rapidly is to trigger this bug within seconds. :-) 
I encountered this by accident, just by changing baudrate/ctsrts once or twice. So it is not a hypothetical issue stemming from rapid changing, it happens at random, but with low-ish probability. 
@arfoll : agree with you about this probably being a kernel issue. 
It struck me after posting too, that users (esp non-root ones) should not be able to cause what this does. Since both IMX UART and FTDI USB serial both have this problem, it is probably related to the kernel tty layer or something. Higher up in the kernel than the physical device drivers.

Considering that stty can do the same thing without problems, means there is a chance to find a workaround.