eclipse-mraa/mraa #737: BeagleBone Black - Examples work but linking mraa with my own file doesn't

ncrastanaren, 2017-05-18:
Seems like some weird linking issue is happening here. 

If I build the examples and run `mraa-gpio list`, I get the expected list of pins. If I take that same mraa-gpio.c file and build with `gcc mraa-gpio.c -lmraa -o mraa-gpio` and run it the same way, no errors are thrown, but I also get no output. 

If I put a print in the `mraa_gpio_init` function, right at the beginning, then do a make install, I get the printout when I run the example you guys build, but mine doesn't show a printout. 

Any idea what might be going wrong here? I'm puzzled.

arfoll, 2017-05-18:
likely you're not installing or linking against the right version of the lib. Check `ldd mraa-gio` and check with what you installed. Likely you have a mraa in /usr/lib (default system one) and another in /usr/local/lib (the one you installed). Cmake (like autotools) by default uses the prefix path /usr/local not /usr. You can change this by setting a cmake build option, check docs/building.md.

arfoll, 2017-05-18:
just to finish the explanation mraa's examples link against the build of mraa in the build/ directory so if you check them with `ldd` they have the full RPATH. This gets removed on `make install`

ncrastanaren, 2017-05-18:
@arfoll Thanks for the reply. I figured something like that was happening. Have it working now.

closed 2017-05-18 by ncrastanaren