eclipse-mraa/mraa #738: jsonplatform.c: potential pin label truncation

alext-mkrs, 2017-05-18:
While analyzing SonarQube findings for mraa I've come across an interesting place (not flagged by Sonar, BTW), namely [line 217 of `jsonplatform.c`](https://github.com/intel-iot-devkit/mraa/blob/master/src/json/jsonplatform.c#L217),

```c
temp_string = json_object_get_string(jobj_temp);
// set the gpio label
strncpy(board->pins[pos].name, temp_string, 8);
```

where we seem to do several suspicious things at once:

1. Copy exactly 8 symbols, while `name` in `mraa_pininfo_t` is declared as `name[MRAA_PIN_NAME_SIZE]` (for the non-PeripheralManager case) where `MRAA_PIN_NAME_SIZE` equals 12. There is no mention in the docs of what the max length of the label is, so we potentially truncate otherwise valid labels (9-12 symbols) and don't null-terminate all ofthem properly;
1. For PeripheralManager `.name` is declared as `char *name` and we don't seem to allocate the memory for it anywhere before copying. I'm not sure if this codepath is at all relevant for AT?

@Hbrinj could you please help me understand if there's any specific idea behind those or we need to fix that?

Hbrinj, 2017-05-18:
Hey Alex, currently finishing exams so I don't have time to take a look. Is it okay if I get back to you after the 23rd?

alext-mkrs, 2017-05-19:
Sure, no problem :-)

arfoll, 2017-05-28:
@Hbrinj update?

alext-mkrs, 2017-06-03:
@Hbrinj, so have you been able to look into this? If not, when?

Hbrinj, 2017-06-03:
Hey guys, I have not, it's on my list starting Monday. I'll give you an update by eod Monday

Hbrinj, 2017-06-05:
So I believe, when this was being done, @arfoll and I had been talking about how long the name should be and an arbitrary number was chosen.  However, due to how long ago this was, I cannot be certain. You seem to be correct @alext-mkrs, it should take MRAA_PIN_NAME_SIZE-1 characters and set the last character as null. This by all intensive purposes is a mistake of mine.

As for point 2, I don't believe I've worked with it. But i would assume from [Line 641 in jsonplatform.c](https://github.com/intel-iot-devkit/mraa/blob/79da382832e5146700fbda820e10d3412bdc63aa/src/json/jsonplatform.c#L641) that it does not matter, no?

alext-mkrs, 2017-06-05:
Ok, got it, thanks, then that would be something up for fixing.

Nope, that line deals with allocating memory for the board structure itself, while the thing I'm after is the `name` member of the `pininfo_t` structure, [here](https://github.com/intel-iot-devkit/mraa/blob/master/include/mraa_internal_types.h#L321). If AndroidThings+JSON platform definition is a valid use case, we don't seem to allocate memory for `name` in this case, so `strncpy` will basically overwrite memory.

commit d544e3c2 (2017-06-21): jsonplatform.c: fix potential segfault at pin label processing

Also update docs to reflect the max label length.

Fixes #738.
closed 2017-06-21 by arfoll