eclipse-mraa/mraa #742: BeagleBone Black - SPI does not work unless config-pin is manually called for each pin

ncrastanaren, 2017-05-24:
Hi guys,

Trying to get SPI up and running on the BBB. I've got a simple loopback example with SPI-D0 connected to SPI-D1:
```c
#include <mraa.h>
#include <stdint.h>
#include <stdio.h>

int main(){

 mraa_spi_context spi;
    spi = mraa_spi_init(0);
    printf("Hello, SPI initialised\n");
    uint8_t data[] = { 0x00, 100 };
    uint8_t* recv;
    while (1) {
        int i;
        for (i = 90; i < 130; i++) {
            data[1] = i;
            recv = mraa_spi_write_buf(spi, data, 2);
            printf("Writing -%i", i);
            printf("RECIVED-%i-%i\n", recv[0], recv[1]);
            usleep(100000);
        }
        for (i = 130; i > 90; i--) {
            data[1] = i;
            recv = mraa_spi_write_buf(spi, data, 2);
            printf("Writing -%i", i);
            printf("RECIVED-%i-%i\n", recv[0], recv[1]);
            usleep(100000);
        }
    }
}
```

This example does not work (receives 0's back) unless I run:
```bash
config-pin P9_17 spi
config-pin P9_18 spi
config-pin P9_21 spi
config-pin P9_22 spi
```

My thought is that the spi module in mraa should take care of setting the pins properly. Is this expected?

```
debian@beaglebone:~$ uname -a
Linux beaglebone 4.4.54-ti-r93 #1 SMP Fri Mar 17 13:08:22 UTC 2017 armv7l GNU/Linux
```

Thanks,
Nick

ncrastanaren, 2017-05-25:
Note that this is only an issue with the universal cape.

arfoll, 2017-05-28:
So as you mentioned in #746 it looks like our pin mapping does not work with newer kernels like 4.4.x :/ It indeed should work as you describe.

icis4, 2019-02-25:
uname -a
Linux beaglebone 4.14.94-ti-r93 #1 SMP PREEMPT Tue Jan 29 18:10:13 UTC 2019 armv7l GNU/Linux

sudo config-pin P9_17 spi_cs
sudo config-pin P9_22 spi_sclk
sudo config-pin P9_21 spi
sudo config-pin P9_18 spi

alext-mkrs, 2019-03-02:
I think that unfortunately no one of the current active contributors has a BBB available, so we'd gladly review a PR with the fix from anyone who's up to that. Let me add a lable for that.

pixsysedp, 2019-06-20:
Is there any news about SPI mraa support on a recent kernel in  BeagleBone Black?
In my code I try to open SPI device using:
`spi = mraa_spi_init(SPI_BUS);` 

I'm usind  SPI_BUS = 2 but it's wrong. In python I can open the device with:

`from Adafruit_BBIO.SPI import SPI`
`import time`
`#https://adafruit-beaglebone-io-python.readthedocs.io/en/latest/SPI.html`
`spi = SPI(2,1)`
`spi.msh = 100000`
`print(spi.xfer2([1, 2, 3, 4, 5, 6]))`
`spi.close()`

Onmy board I have:
`ls -lah /sys/class/spidev/`
`lrwxrwxrwx  1 root root 0 Jun 20 12:37 spidev2.0 -> ../../devices/platform/ocp/481a0000.spi/spi_master/spi2/spi2.0/spidev/spidev2.0`
`lrwxrwxrwx  1 root root 0 Jun 20 12:37 spidev2.1 -> ../../devices/platform/ocp/481a0000.spi/spi_master/spi2/spi2.1/spidev/spidev2.1`