eclipse-mraa/mraa #743: mraa with upstream kernel on Galileo and IOT2000

jan-kiszka, 2017-05-26:
This is about starting a discussion what is missing to move forward towards always-fresh upstream kernels with the Galileo Gen 2 board and our derived IOT2000 series. Maybe this affects even more Intel boards that still depend on vendor BSPs.

Current issues:
* IIO ADC driver for TI ACC108S102 => merge pending, driver should make it for 4.13
* dynamic GPIO numbering => likely fixable in mraa, already working on a patch that uses firmware_node/path (ACPI paths) as stable anchor
* fast GPIO access via uio driver => already optional, but may either be replace with generic  GPIO_GET_LINEHANDLE_IOCTL or - if that is still too slow (do you know use cases?) - /dev/mem hack (fragile in case some pins of a register are used otherwise, just like with uio already)
* non-upstream GPIO pin drive => still unclear, examining
* anything missing?

Before going to LKML/linux-gpio with discussions on the GPIO drive feature - seems like it is a userspace API to pinctrl, but I'm still learning about this -, I'd like to understand the requirements and use cases.

First a detail question on mraa_intel_galileo_gen2_gpio_mode_replace (likely for @tingleby): Why this mapping of strong -> hiz, pullup -> strong, hiz -> nop? I guess the latter is because hiz implies input, and in that case the PCAL9535 disables pulling implicitly. But the other cases remain unclear.

Then: Setting the pull mode of a GPIO pin is an official mraa API, right? Could we disable this feature for upstream-based boards without breaking too many use cases? Or is there a concept available to build on existing upstream features or add new ones to restore this? I'm not talking about the static ACPI table extensions like done in https://github.com/westeri/meta-acpi, because they cannot be reconfigured once applied.

arfoll, 2017-05-31:
Ah the gen2 galileo :) Brings back memories.

So GPIO_GET_LINEHANDLE_IOCTL access will likely never be implemented, I'd rather move to the gpio chardev that does similar things and is newer/supported etc... See the other issue on the topic.

The pindrive stuff was pretty specific to the driver, maybe we need a check to avoid doing this at all if we detect a newer kernel. At the same time, I'm not too bothered if we break support for galileo with super old kernels so if this stuff is no longer needed we can remove it. @tingleby do you remember why the mapping is done this way?

And yes the pull/mode is a mraa feature, but since it's overriden by the gen2 platform config already I suggest you just add a check for the kernel version and then based on this just disable/change the functionality in `mraa_intel_galileo_gen2_gpio_mode_replace`. That way we don't break the ancient stuff as well as work on newer stuff :)

jan-kiszka, 2017-06-01:
> So GPIO_GET_LINEHANDLE_IOCTL access will likely never be implemented, I'd rather move to the gpio chardev that does similar things and is newer/supported etc... See the other issue on the topic.

To my understanding, GPIO_GET_LINEHANDLE_IOCTL is what is  applied on the gpio chardev. See also linux/tools/gpio/gpio-utils.c. There are no other chardevs implemented by gpiolib.

> [...]
> And yes the pull/mode is a mraa feature, but since it's overriden by the gen2 platform config already I suggest you just add a check for the kernel version and then based on this just [...]

This is what I'm already doing, and it seems to work (several tests are pending, my test hardware didn't arrive yet). I will share patches soon.

tingleby, 2017-06-01:
Let me try and remember why its like this.

So the replace function configures the mux/level-shift device in-between the actual gpio line and what ends up coming out of the dev board. Although these usages may have just been what worked on whatever driver versions I was on at the time.

jan-kiszka, 2017-06-02:
Meanwhile I uploaded the patches, see https://github.com/siemens/mraa/commits/jan/queue

jan-kiszka, 2017-06-13:
Any comments on my commits? Would it already make sense to send them as PR?

I didn't test all I/O features abstracted by mraa yet, primarily because my test shields are still stuck in customs clearing. But I suspect the rest should be changes on top.