eclipse-mraa/mraa #745: mraa npm package bundles unnecessary code

dnoliver, 2017-05-26:
mraa package has several unnecessary files:

```
dnoliver@dnoliver-vm:~/Git/test/node_modules/mraa$ ls -a
drwxrwxr-x 13 dnoliver dnoliver   4096 May 26 08:41 .
drwxrwxr-x  3 dnoliver dnoliver   4096 May 26 08:41 ..
drwxrwxr-x  3 dnoliver dnoliver   4096 May 26 08:41 api
-rw-rw-r--  1 dnoliver dnoliver   1863 May 12 05:22 binding.gyp
drwxrwxr-x  3 dnoliver dnoliver   4096 May 26 08:41 build
-rw-rw-r--  1 dnoliver dnoliver   1511 Apr  2 11:22 .clang-format
drwxrwxr-x  4 dnoliver dnoliver   4096 May 26 08:41 cmake
-rw-rw-r--  1 dnoliver dnoliver   9832 May 12 05:22 CMakeLists.txt
-rw-rw-r--  1 dnoliver dnoliver   4269 Apr  2 11:22 CONTRIBUTING.md
-rw-rw-r--  1 dnoliver dnoliver   1087 Apr  2 11:22 COPYING
drwxrwxr-x  2 dnoliver dnoliver   4096 May 26 08:41 docs
-rw-rw-r--  1 dnoliver dnoliver 102142 Apr  2 11:22 Doxyfile.in
-rw-rw-r--  1 dnoliver dnoliver 101475 Apr  2 11:22 Doxyfile.java.in
drwxrwxr-x  2 dnoliver dnoliver   4096 Apr  2 11:22 doxygen2jsdoc
-rw-rw-r--  1 dnoliver dnoliver   6087 Apr  2 11:22 DoxygenLayout.xml
drwxrwxr-x  7 dnoliver dnoliver   4096 May 26 08:41 examples
-rw-rw-r--  1 dnoliver dnoliver    118 Apr  2 11:22 .gitmodules
drwxrwxr-x  2 dnoliver dnoliver   4096 May 26 08:41 imraa
drwxrwxr-x 11 dnoliver dnoliver   4096 May 26 08:41 include
drwxrwxr-x  3 dnoliver dnoliver   4096 May 26 08:41 jsstub
-rw-rw-r--  1 dnoliver dnoliver   2305 May 26 08:41 package.json
-rw-rw-r--  1 dnoliver dnoliver   2153 May 12 05:22 READMEFIRST
-rw-rw-r--  1 dnoliver dnoliver   5460 May 12 05:22 README.md
drwxrwxr-x 21 dnoliver dnoliver   4096 May 26 08:41 src
drwxrwxr-x  3 dnoliver dnoliver   4096 May 26 08:41 tests
-rw-rw-r--  1 dnoliver dnoliver   2464 May 12 05:22 .travis.yml
``` 

`package.json` do not have a "file" property, so it is adding everything to the package.

arfoll, 2017-05-28:
Do we need a 'file' property for what then? Just the compiled stuff?

dnoliver, 2017-05-28:
@arfoll I am not aware of how you are creating the npm package to publish, but the "standard" way is to just do `npm publish` and let npm handle the packaging for you. You can check what is going to be publish by running `npm pack` in your code and check what is included in the tar.gz file. By using the [package.files](https://docs.npmjs.com/files/package.json#files) property, you tell npm which things should be included in the package to be published. 

So I should take a look at the current job that package and publish mraa to npm first, but if you are just using `npm publish`, just modifying the package.json template could be enough.

arfoll, 2017-05-30:
check docs/npm.md and the package.json is generated from package.json.cmake. I guess the problem is we just use the root directory to pack everything instead of a new dir. There was a reason for that but I can't remember. Gues I need to have another look

arfoll, 2017-05-30:
Seems like if I add:
```
"main": [ "./mraa.node" ]
```
to package.json.cmake I can get it to build and seems like it only then copies the mraa.node file. Does that sound about right?

dnoliver, 2017-05-30:
The "main" field just tell npm that that file is the entrypoint, so if you require() the package, that file will be imported. But the only way I know to tell npm pack which files to include in the package is using the "files" field (with a combination with .npmignore file if needed)