eclipse-mraa/mraa #746: Beaglebone Black - PWM does not work

ncrastanaren, 2017-05-26:
On the beaglebone black, your code is looking for a path like `/sys/class/pwm/pwm6/export`, when the newer kernels instead expose `/sys/class/pwm/pwmchip6`. 

Comparing your library with the adafruit bbio library, your mappings seem to be different as well.

Is there any chance you could tell me what image was on the beaglebone this stuff was tested on?

arfoll, 2017-05-28:
You probably want an answer from: @svlstriker @michael-ring. I'm not sure what they used however docs/beaglebone.md indicate a 3.8.13 debian image.

jan-kiszka, 2017-05-28:
That code may have been based on some old vendor kernel. Upstream Linux always used the pwmchipN/pwmX hierarchy, so did the pwm subsystem of mraa. Time to fix the board code, I would say. There is no point in hunting down the historic system this once worked on - if it did so at all.

jan-kiszka, 2017-05-28:
More precisely, upstream gained pwn sysfs with 3.11 in 2013 - which means 3.8 was nothing official. The old problem of relying on downstream APIs.

arfoll, 2017-05-28:
Unfortunately I don't even have a bbb. That code comes from the
contributors above. I'm surprised though because phytec have been
advertising mraa support on their BBB clone surely they're not using an
ancient 3.8 kernel... Here's to hoping!

ncrastanaren, 2017-05-30:
Is there support for 4.x kernels with other boards?

I am planning on trying to fix the BBB code based on the Adafruit library. I will make a PR when/if I get something usable.

arfoll, 2017-05-30:
For sure, Intel Joule & Minnowboard Turbot definately support 4.x kernels. On the arm side I think the pi3 kernel is pretty new but dont have one to hand to be 100%

Chuckduey, 2018-10-02:
I am currently working with BBB and PWM.  I created a workaround using low level kernel calls.  Not a very elegant solution.  I would rather get an MRAA version that works than to use my ugly work-around.  It looks like PR #273 may be a fix for this.   Any status on getting a new PR or update for the PWM problem?

nick-atmosphereiot, 2018-10-02:
I ended up writing a solution, but I do not recall if I opened up a PR:
https://github.com/nick-atmosphereiot/mraa/commits/master