eclipse-mraa/mraa #751: Travis build on Docker do not run test for Pyhton3

dnoliver, 2017-05-31:
https://travis-ci.org/intel-iot-devkit/mraa/jobs/237982191

Build Log
```
canning dependencies of target _python3-mraa
[ 92%] Building CXX object src/python/python3/CMakeFiles/_python3-mraa.dir/mraa3PYTHON_wrap.cxx.o
[ 96%] Building C object src/python/python3/CMakeFiles/_python3-mraa.dir/__/mraapy.c.o
[100%] Linking CXX shared module _mraa.so
[100%] Built target _python3-mraa
Test project /usr/src/app/build
No tests were found!!!
```

dnoliver, 2017-05-31:
java target works fine, it execute tests. Node4 and Node5 do not use ctest, npm test run well

alext-mkrs, 2017-06-03:
Is this still a problem?I think @arfoll fixed this in his recent commits (1d37d3b1624ff953beb1d456e2d1a4a3ea1549ec, 131ff46a0d14935571bb6f49ea0b9ca7e3e69c04)..?

dnoliver, 2017-06-05:
@alext-mkrs, This was detected on the last build in Travis for Master branch. I will execute the builds again to confirm

alext-mkrs, 2017-06-05:
I think I could do that, but that build is all green except those that are marked as "can fail". Is that really what you need to do here?

alext-mkrs, 2017-06-05:
Ah, never mind - got it, after re-reading the issue description and looking at logs. So you mean the tests are not being run... Let me run it once again to double-check, but yeah, probably there's some stuff still to do here.

alext-mkrs, 2017-06-05:
Ok, it doesn't run still and I think the reason is within commit 1d37d3b1624ff953beb1d456e2d1a4a3ea1549ec, namely the addition of `if (PYTHON2INTERP_FOUND >>>AND EXISTS "${CMAKE_BINARY_DIR}/src/python/python2/mraa.py"<<<)` part. When cmake runs through these lines, the `make` naturally hasn't run yet, so there are no such files, so it doesn't set the `PYTHON_DEFAULT_INTERP` variable and that causes it not to add tests to the list later on.

```text
-- INFO - swig Version 3.0.11
-- INFO - CXX compiler Version GNU 6.3.1
-- INFO - Using V8 version > 3 so requiring C++11 compiler
-- INFO - install NODE modules to /usr/lib/node_modules/mraa/
-- Could not run tests since python interpreter or python bindings not built <<<<<< here's the error message it throws
-- Configuring done
-- Generating done
-- Build files have been written to: /home/at/mraa/mraa-git/build
```

@arfoll, do we really need those additional checks? Looking at them I don't fully get why those would be beneficial here at all?

alext-mkrs, 2017-06-10:
@arfoll, ping.

arfoll, 2017-06-12:
I spent a little time on it a few weeks ago and I'm not sure why I did that check anymore, there was a reason... I put a `testfix` branch with the removal of those to check. Tell me if that's any better?

alext-mkrs, 2017-06-12:
Thanks for clarification! I see python 3 builds failed for whatever reason, though I can't see why just by inspection - at the first glance that change you made in that branch is what's actually needed. I'll check it in more detail tomorrow.

arfoll, 2017-06-12:
Yeah I have something not committed yet, actually it doesn't work. Essentially it doesn't work this way, basically we need to either build everything in the test suite and/or have a seperate PYTHON3/PYTHON2 target to do this. At least I cannot see a way of doing it otherwise. Basically I do not know if _python3-mraa was built when you run `make test` I can only find out if it's been configured (the dir in build/ exists) and if python3 lib/interp has been found. I can make the test depend on _python3-mraa but that's not enough to actually get them built and if you have python2 I'll still prefer that.

This is probably more complexity than it's worth I suggest that we use a flag to force python3 tests and enable it in travis for python3 builds. @alext-mkrs @dnoliver you guys ok with that?

alext-mkrs, 2017-06-13:
Yes, I'm fine with that. It's interesting why target dependency didn't do the trick, but flag works and should be easy to change if we discover other - better - ways in the future.

arfoll, 2017-06-13:
it seems to be that `make test` doesn't force deps to get built, if you run a full `make` then `make test` it will work but then unless you don't have python2/python2-dev you'll always default to python2 tests.

dnoliver, 2017-06-13:
I am ok with a flag to enable Python3 tests!

arfoll, 2017-06-13:
so let's see how e66c8568b45e89d51112265ea101bfdb753977fa fares :)

alext-mkrs, 2017-06-13:
Looks like it didn't fully help it. I'm currently looking into this too. On my dev machine it seems to work fine, but on Travis for whatever reason PYTHONPATH is being set to python2 even when I force the python3tests flag. I've added some debug into cmakefiles and checking what's going on there.

dnoliver, 2017-06-13:
take into account that Travis is running `make  _python3-mraa && ctest --output-on-failure` to run the tests. If you do `make && ctest --output-on-failure` you will have 100% test pass, but it will be because make generates python2 libs, so the test use that.

alext-mkrs, 2017-06-13:
yep, I know that and it actually properly uses python3 on my machine - and passes.

alext-mkrs, 2017-06-13:
```shell
$ make _python3-mraa && ctest --verbose
<cut>
test 1
    Start 1: py_general

1: Test command: /usr/bin/python3 "/home/at/mraa/mraa-git/tests/general_checks.py"
1: Environment variables: 
1:  PYTHONPATH=/home/at/mraa/mraa-git/build/src/python/python3
1: Test timeout computed to be: 9.99988e+06
1: .
1: ----------------------------------------------------------------------
1: Ran 1 test in 0.000s
1: 
1: OK
1: Version is: v1.7.0-40-ga103051
1/3 Test #1: py_general .......................   Passed    0.12 sec
<cut>
```

dnoliver, 2017-06-13:
do you have docker and docker compose in you development machine? if you do, ``docker-compose build python3 && docker-compose run python3` should replicate the issue

alext-mkrs, 2017-06-13:
I don't (yet), but I found it. `USEPYTHON3TESTS` was not passed to the CMake run in the Dockerfile, only declared as arg. That caused it to always enter the Python 2 clause of the selection if in the test dir's CMakeLists.txt. Patch is on the way.

alext-mkrs, 2017-06-13:
Here it comes: https://travis-ci.org/intel-iot-devkit/mraa/builds/242557274

dnoliver, 2017-06-13:
Hey it works! :D now python3 test runs! 

It is only having problems with -DBUILDARCH=MOCK (https://travis-ci.org/intel-iot-devkit/mraa/jobs/242557283) apparently it is still using python2 there

alext-mkrs, 2017-06-13:
Yeah, mock cmake config needs to be updated to use `PYTHON_DEFAULT_INTERP` we've introduced instead of `_EXECUTABLE`, on it.

alext-mkrs, 2017-06-13:
https://travis-ci.org/intel-iot-devkit/mraa/builds/242566929. When we stablize it, we really need to think about building the container once and then just using it (updating only sources) - all those package updates and tool installs take like 3/4 of the build time...

alext-mkrs, 2017-06-13:
Ok, it works ok now. @arfoll, please note I've additionally fixed unrelated something, that looks like a typo to me, though apparently it was harmless as it was there for a while and nothing broke, this commit in the `testfix` branch: ef79bc37862738e051aed3d9e17b82da87473776

dnoliver, 2017-06-13:
@alext-mkrs you are right. 

We can leverage the existing implementation using Travis Build Stages (https://docs.travis-ci.com/user/build-stages) to prepare the docker image to use for every plan. Travis also recommends using DockerHub as a cache for it. So we can just build the image a few times, publish it in docker hub, and then just pulling the image to build and run tests

alext-mkrs, 2017-06-13:
Right, that's exactly what I'm thinking about here. Though not sure Stages would be useful in our case - as we don't want to build it every tim, even in a staged manner, i.e. before the actual mraa build - because that would be effectively the same thing we do now, if that's what you mean.

I'm thinking about something more like just building it once in a while when our tools/dependencies change and then during mraa build just pulling it off of Docker Hub and only `run`ning it vs. `build`ing and `run`ning we do now. Though `docker-compose` brings in certain complication here as we effectively have several containers, one for each service... We'll need to think about the best way to approach this, for sure :)

alext-mkrs, 2017-06-15:
@arfoll, would you want to do anything else here? I think we have this sorted out now and it's just a matter of squashing it and getting into `master`. I can do that if you want.

alext-mkrs, 2017-06-17:
@arfoll, ping. It's a bit inconvenient to merge these fixes all the time when working on other stuff, to have tests working :)

arfoll, 2017-06-17:
I'm not sure what you want me to merge. It doesn't fix the issue at all :/
tbh at this point I'm thinking just disable python3 tests. They add very
little value anyways

alext-mkrs, 2017-06-18:
Hmm, the "testfix" branch at least does fix tests not being run at all (that original problem) and per my tryouts, does run tests with python3 with the flag and with python2 - without it (the one that came up during fixing). Are you saying it doesn't for you?

That's with both python 2 and 3 installed on my dev machine, haven't tried one or the other.

alext-mkrs, 2017-06-18:
Or... have you by any chance not seen my last couple of commits in that branch, which fixed the ARG not being passed to Docker?

arfoll, 2017-06-19:
I did not, my bad!

Alot of magic rebasing and I think I didn't screw it up! I merged testfix with a few less commits that are a bit more logical.

arfoll, 2017-06-19:
Thanks to everyone involved, marking this as closed!

closed 2017-06-19 by arfoll

alext-mkrs, 2017-06-20:
Yeah, commits look good :) Thanks!