eclipse-mraa/mraa #753: Intel Joule SPI not working

AkshayBhagat, 2017-06-01:
<h1>Platform</h1>
Board:Intel Joule 570X Expansion Board<br>
<h3>OS:</h3> uname -a<br> 
Linux scout-570x-DVT3 4.4.0-1000-joule #0+joule12-Ubuntu SMP PREEMPT Fri Jan 6 04:28:41 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux<br>
lsb_release -a<br>
No LSB modules are available.<br>
Distributor ID:	Ubuntu<br>
Description:	Ubuntu 16.04.1 LTS<br>
Release:	16.04<br>
Codename:	xenial<br>
<h3>MRAA</h3>
mraa-gpio version
Version v1.7.0-11-g47c21e0 on INTEL JOULE EXPANSION<br>
<h1>Program:</h1>
<h3>Source</h3>
#include "mraa.h"<br>
#include <unistd.h><br>
#include <stdint.h><br>
#include<iostream><br>
int main(int argc, char** argv)<br>
{
    mraa_init();<br>
    //! [Interesting]<br>
    mraa_spi_context spi;<br>
    spi = mraa_spi_init(0);<br>
    printf("Hello, SPI initialised\n");<br>
    std::cout<<"spi init"<<std::endl;<br>
    uint8_t data[] = { 0x00, 0x32 };<br>
    uint8_t* recv;<br>
    std::cout<<"assigning  data"<<std::endl;<br>
    int i=90;<br>
    data[1] = i;<br>
    std::cout<<"spi write coming up"<<std::endl;<br>
    recv = mraa_spi_write_buf(spi, data, 2);<br>
    if (recv==NULL)<br>
        std::cout<<"error in writing"<<std::endl;<br>
   printf("Writing -%i\n", i);<br>
    //printf("RECIVED-%i-%i\n", recv[0], recv[1]);<br>
    usleep(100000);<br>
    //! [Interesting]<br>
}<br>
<h3>Output</h3>
Hello, SPI initialised<br>
spi init<br>
assigning  data<br>
spi write coming up<br>
error in writing<br>
Writing -90<br>

<h1>Debug</h1>
journalctl -t libmraa<br>
-- Logs begin at Thu 2017-04-13 12:04:32 EDT, end at Thu 2017-06-01 18:28:10 EDT. --
Jun 01 18:18:37 scout-570x-DVT3 libmraa[2837]: libmraa version v1.7.0-11-g47c21e0 initialised by user 'scout' with EUID 1001<br>
Jun 01 18:18:37 scout-570x-DVT3 libmraa[2837]: Adding i2c bus found on i2c-4 on adapter i2c_designware.0<br>
Jun 01 18:18:37 scout-570x-DVT3 libmraa[2837]: Adding i2c bus found on i2c-9 on adapter i2c_designware.5<br>
Jun 01 18:18:37 scout-570x-DVT3 libmraa[2837]: Adding i2c bus found on i2c-10 on adapter i2c_designware.6<br>
Jun 01 18:18:37 scout-570x-DVT3 libmraa[2837]: Adding i2c bus found on i2c-5 on adapter i2c_designware.1<br>
Jun 01 18:18:37 scout-570x-DVT3 libmraa[2837]: Adding i2c bus found on i2c-6 on adapter i2c_designware.2<br>
Jun 01 18:18:37 scout-570x-DVT3 libmraa[2837]: libmraa initialised for platform 'INTEL JOULE EXPANSION' of type 13<br>
<b>Jun 01 18:18:37 scout-570x-DVT3 libmraa[2837]: spi: Failed opening SPI Device. bus:/dev/spidev32765.0. Error 2 No such file or directory<br>
Jun 01 18:18:37 scout-570x-DVT3 libmraa[2837]: spi: write_buf: context is invalid<br></b>
Jun 01 18:24:50 scout-570x-DVT3 libmraa[3049]: libmraa version v1.7.0-11-g47c21e0 initialised by user 'scout' with EUID 1001<br>
Jun 01 18:24:50 scout-570x-DVT3 libmraa[3049]: Adding i2c bus found on i2c-4 on adapter i2c_designware.0<br>
Jun 01 18:24:50 scout-570x-DVT3 libmraa[3049]: Adding i2c bus found on i2c-9 on adapter i2c_designware.5<br>
Jun 01 18:24:50 scout-570x-DVT3 libmraa[3049]: Adding i2c bus found on i2c-10 on adapter i2c_designware.6<br>
Jun 01 18:24:50 scout-570x-DVT3 libmraa[3049]: Adding i2c bus found on i2c-5 on adapter i2c_designware.1<br>
Jun 01 18:24:50 scout-570x-DVT3 libmraa[3049]: Adding i2c bus found on i2c-6 on adapter i2c_designware.2<br>
Jun 01 18:24:50 scout-570x-DVT3 libmraa[3049]: libmraa initialised for platform 'INTEL JOULE EXPANSION' of type 13<br>
Jun 01 18:28:04 scout-570x-DVT3 libmraa[3115]: libmraa version v1.7.0-11-g47c21e0 initialised by user 'scout' with EUID 1001<br>
Jun 01 18:28:04 scout-570x-DVT3 libmraa[3115]: Adding i2c bus found on i2c-4 on adapter i2c_designware.0<br>
Jun 01 18:28:04 scout-570x-DVT3 libmraa[3115]: Adding i2c bus found on i2c-9 on adapter i2c_designware.5<br>
Jun 01 18:28:04 scout-570x-DVT3 libmraa[3115]: Adding i2c bus found on i2c-10 on adapter i2c_designware.6<br>
Jun 01 18:28:04 scout-570x-DVT3 libmraa[3115]: Adding i2c bus found on i2c-5 on adapter i2c_designware.1<br>
Jun 01 18:28:04 scout-570x-DVT3 libmraa[3115]: Adding i2c bus found on i2c-6 on adapter i2c_designware.2<br>
Jun 01 18:28:04 scout-570x-DVT3 libmraa[3115]: libmraa initialised for platform 'INTEL JOULE EXPANSION' of type 13<br>
Jun 01 18:28:10 scout-570x-DVT3 libmraa[3116]: libmraa version v1.7.0-11-g47c21e0 initialised by user 'scout' with EUID 1001<br>
Jun 01 18:28:10 scout-570x-DVT3 libmraa[3116]: Adding i2c bus found on i2c-4 on adapter i2c_designware.0<br>
Jun 01 18:28:10 scout-570x-DVT3 libmraa[3116]: Adding i2c bus found on i2c-9 on adapter i2c_designware.5<br>
Jun 01 18:28:10 scout-570x-DVT3 libmraa[3116]: Adding i2c bus found on i2c-10 on adapter i2c_designware.6<br>
Jun 01 18:28:10 scout-570x-DVT3 libmraa[3116]: Adding i2c bus found on i2c-5 on adapter i2c_designware.1<br>
Jun 01 18:28:10 scout-570x-DVT3 libmraa[3116]: Adding i2c bus found on i2c-6 on adapter i2c_designware.2<br>
Jun 01 18:28:10 scout-570x-DVT3 libmraa[3116]: libmraa initialised for platform 'INTEL JOULE EXPANSION' of type 13<br>

<h1>Comments</h1>
Do I need to enable spidev for libmraa to work. If so how do I enable it?<br>

alext-mkrs, 2017-06-03:
First of all I'd suggest you to run your app as root (or using sudo) as mraa needs that to work properly. Let's see if that makes any difference. Plus, I'd suggest you to update mraa to the latest version, there have been fixes for Joule recently.

aj-ptw, 2017-06-08:
having same issue

alext-mkrs, 2017-06-08:
@akshaybhagat, th log you posted states the effective user id of the process is 1001 (user "scout"), so it doesn't look like sudo was used.

As for the version - 1.7.0 is the latest, right, but then if you check the specific commit hash (that hex string after version), you'll notice it's behind the current master branch. What I'm suggesting here is either to install the latest dev release (built daily) or build from sources - to make sure you have all the latest fixes.

I also see you have Ubuntu installed, have by chace tried with the default Ref OS for IoT?

In general, if no one tackles this sooner, I plan to look into this over the upcoming weekend.

Propanu, 2017-06-08:
Hi @AkshayBhagat, as of now SPI only works on the Ref-OS-IoT image.

aj-ptw, 2017-06-08:
@propanu that's false 

See https://communities.intel.com/thread/115016
And https://communities.intel.com/thread/114441

What are you using?

Propanu, 2017-06-08:
[This](https://github.com/ostroproject/ostro-os-xt/releases/tag/v1.0.1). Ref-OS used to be called Ostro. Wouldn't be surprised if SPI was broken in one of the newer releases.

aj-ptw, 2017-06-08:
@Propanu are you suggesting to load that Ostro version, before is was named Ref-OS to the Intel Joule because it was in a known working state? Thanks so much for your help!

Propanu, 2017-06-09:
Yes, although updating the MRAA library on Ostro is a bit of a hassle. Most likely you'll have to compile it from scratch. Looking into it to see if there's a way to fix it for the new Ref-OS images too.

alext-mkrs, 2017-06-10:
So by now you probably understand this is not mraa's problem, but an OS-level one. I'm currently looking into this and I think I've found the specific reason, running some tests to validate that. Will post back when done.

alext-mkrs, 2017-06-11:
I've found at least a solid interim solution, but as long as this is an OS-level thing and I see @aj-ptw has already submitted an issue to ref-os-iot's repo (https://github.com/01org/ref-os-iot/issues/2) - let me continue the discussion there and close this one.

On a side note - @arfoll or @tingleby, an "OS issue" label would come in handy for marking such cases, similarly to "hardware issue" we already have. I'm not able to create one, so if you too think it could be useful and have a spare minute - please create it.

closed 2017-06-11 by alext-mkrs

aj-ptw, 2017-06-11:
@alext-mkrs thanks so much for the follow up.