eclipse-mraa/mraa #754: Is this lib the right choice for Beagle Bone Black and kernel 4.4.x programming?

mgiaco, 2017-06-03:
Hi,

At my daily work I use devices based on ARM7, Cortex M3/M4 and AVR. I mostly programming these devices in plain C. 

So no for a little private project I would like to use a linux board with python. So therefore I found this lib.  I would like realize a little project with some sensors and actors. I need I2C, SPI and UART and also some normal IO´s. I would like to use a Beaglebone Black for all that work.

And know I am wondering if I am right here? I see that there are some issues around for BBB so I am not sure. I also would like to use a yocto based image but that's not the problem. The image image is based on the latest kernel 4.4.x

many thanks

alext-mkrs, 2017-06-03:
Depending on what exactly you're trying to do. Generally, BBB part of the code just needs a little love (as non of the regular contributors has this board) and that's being currently done within PR #747.

mgiaco, 2017-06-03:
Okay that sound no so good. So this lib is only recommended for Intel boards - right?

nrcrast, 2017-06-04:
@mgiaco I've been working on the BBB in my fork. I've got all of the peripherals (GPIO, SPI, I2C, ADC, PWM, UART) working/tested, so you could always give it a shot.

The Adafruit BBIO library is also really good if you don't care too much about a cross platform library. It's got C/C++/Python hooks.

mgiaco, 2017-06-04:
@nrcrast many thanks.  If I use python can I use python 3.5 because I would like to use asyncio for some stuff. The Adafruit BBB library is only for python 2.7 what I have read. 

If I use C I found libuv which seems to be also a great choice. 

So I am not sure, do know which way is faster?

mgiaco, 2017-06-05:
@nrcrast So I will give your branch a try. So as I say before I am quite new to linux dev. So I taught first i clone your branch and compile it on the BBB normal Debian Image. I use the latest image and I follow this document here http://www.elec-tron.org/?p=996. Is that the right way to go?
Or do you know any guide how to start the right way? 
many thanks

nrcrast, 2017-06-05:
I am using:
[Debian 8.7 IOT Edition](https://debian.beagleboard.org/images/bone-debian-8.7-iot-armhf-2017-03-19-4gb.img.xz)

Clone my aio_bbb branch:
https://github.com/nrcrast/mraa/tree/aio_bbb

Then follow the build instructions from the [docs](https://github.com/intel-iot-devkit/mraa/blob/master/docs/building.md). In short:
1.) navigate to mraa directory
2.) `mkdir build`
3.) `cd build`
4.) `cmake .. -DBUILDSWIGPYTHON=OFF -DBUILDSWIGNODE=OFF -DCMAKE_INSTALL_PREFIX:PATH=/usr`
5.) `make`
6.) `sudo make install`

You should be all set from there.

mgiaco, 2017-06-05:
many thanks but when I would like to try the C way and the python way I should use
```
cmake .. -DBUILDSWIGPYTHON=ON -DBUILDSWIGNODE=ON -DCMAKE_INSTALL_PREFIX:PATH=/usr
```
right?

mgiaco, 2017-06-05:
ah i think I only need
``` cmake` .. -DCMAKE_INSTALL_PREFIX:PATH=/usr  ```

arfoll, 2017-06-05:
You always build the C lib. The python bindings are essentially added onto
and depend on the base lib. If you have swig installed and python-dev cmake
should do "the right thing". Read docs/building.md to learn more

mgiaco, 2017-06-05:
@nrcrast  many thanks got it to build and it seems to work.  When I use mraa-gpio list I got 92 pins listed.  So I do not load any special device tree because first I had to find out all that device tree stuff :-).

But is there a way to control the onboard leds? In the Adafruit_BBB lib I can use USR0..USR3

nrcrast, 2017-06-05:
Not currently. They are in a different place in the sys/class interface and don't really fit in with any current constructs in mraa. You can write your own code to control them, though:

http://robotic-controls.com/book/export/html/69

alext-mkrs, 2017-06-10:
Now that updates from Nick are merged into master, I think we can close this one.

closed 2017-06-10 by alext-mkrs