eclipse-mraa/mraa #758: [RFC] Reverse SPI bus numbering on 5xx

alext-mkrs, 2017-06-11:
Hi @arfoll (as the author of that piece of code), @propanu and @whbruce (as co-maintainers) - in the [5xx board file](https://github.com/intel-iot-devkit/mraa/blob/master/src/x86/intel_joule_expansion.c#L141-L146) we configure SPI buses in the order of increasing OS bus number. Is there any specific reasoning behind this scheme? My proposal would be to reverse the numbering.

#### Reasoning
Given the fact that this platform uses dynamic bus numbering and in such case the OS starts numbering from max and counts downwards, the resulting mraa bus numbers are basically the reverse of what one would expect looking at 5xx pin mapping documents.

E.g. pins marked as `SPI1*` in the [document](https://www.intel.com/content/www/us/en/support/boards-and-kits/intel-joule-kits/000022494.html), belong to mraa's bus number 0 (because it has lower OS number - 32765) and `SPI0*` - to mraa bus number 1 (OS: 32766). I think I've seen some posts from folks confused by this on Intel 5xx forums (though our docs clearly map board pin numbers to mraa bus number, that I give us :)

In addition, per the same document, SPI0 has three chip/slave selects routed to Expansion board headers (0, 1, 2), while we have only 0 in the config. I propose adding those too.

What do you (or anyone else reading this and interested) think about that?

Propanu, 2017-06-11:
@alext-mkrs I agree with the change. In fact, I've also suggested reordering the I2C buses to match the documentation with my #667 PR. I don't mind handling this when I rebase that PR to address @arfoll's comments.

arfoll, 2017-06-12:
In mraa a 'spi' object usually refers to a SPI bus + hardware CS. So generally naming Spi bus 0 '0' doesn't work since really you'd have numerous busses. So I took the approach to just name them as they turned up on the board based on whichever bus + CS (or FS on the datasheet - anyone know the diff between CS/FS/SS?). So that's why bus 1 is mraa bus 0 and not because of the kernel ordering (that just seems to match accidentally more than anything else).

It's not perfect by any means but otherwise you'd make SPI bus 1 actually 2 because you have SPI bus 0 has 2 FS pins (that actually worked at the time, if the others work then let's add them), I suspect changing it to do this would just generate more confusion and you'd still not achieve the 1:1 mapping users would expect.

Propanu, 2017-06-12:
So you're saying having:

| MRAA Bus | SPI Bus |
|:---:|:---:|
| 0 | SPI0.0 |
| 1 | SPI0.1 |
| 2 | SPI0.2 |
| 3 | SPI1.0 |
| 4 | SPI1.1 |

Will be just as confusing, however at least SPI0.0 will really be bus 0. @arfoll I think CS/FS/SS are just different naming conventions for the same pin, functionality wise they "should" be similar but since it's not standardized, devices have a bit of flexibility as to how they use it.

alext-mkrs, 2017-06-12:
Yeah, I'd be in favor of rehauling it, still - that will bring us in line with docs, at least conceptually (mraa bus numbers will increase with spec's ones). As for bus/chip select combinations being a "bus" - which is a good catch - I think we could just add that to docs, with examples. I can do that, along with double-checking whether those other CS pins really work per the spec. But just to make sure, @propanu, you said you want to implement this, along with I2C changes?

AFAIK, CS is Chip Select, SS is Slave Select (typically found in SPI bus markings) and FS is Frame Sync - this one is from I2S, which is frequently implemented in HW by the same high-speed generic serial devices as SPI (e.g. Edison has it like that) and has the same 4-wire scheme.

Propanu, 2017-06-12:
@alext-mkrs, let me know which ones work correctly and I'll map them as listed in the table when I rebase that PR (hope to get to it this week).

arfoll, 2017-06-12:
Yeah actually that's nice. If we can map them like this (and they actually work - this point I'm pretty unsure ;-)) I'm happy to have that change.

@alext-mkrs will have to read more about FS/frame sync, thanks for the info!

alext-mkrs, 2017-06-13:
@arfoll, my pleasure :)

Ok, agreed then - I'll check those buses/chip selects out and post results here.

alext-mkrs, 2017-06-15:
All right, I've tested all available CSes and all work fine (with proper pull-ups and `spidev` binding config). 

As long as I anyway had to make code modifications to test this and I also make docs on SPI a bit more clear, I'm submitting this as a PR right away, @Propanu thanks for offering to do that in any case.

Propanu, 2017-06-15:
That's fine by me. Btw, what exactly do you mean by `spidev` binding config? Can you export a bus config from user space without patching the kernel (rebuilding ref-os-iot)?

alext-mkrs, 2017-06-15:
Not from user space, unfortunately - I did some research on that while invetigating #753 and there seems to be no way for that. There was some [proposal](https://patchwork.kernel.org/patch/1470041/) on LKML, which wasn't accepted due to limited usefulness. Other than that, the API used is kernel-level, so is not usable from userspace.

But you don't really need to rebuild the whole OS image for that - it's possible to build a kernel module, which will do just that, I've built one for me while working on this PR, based on [Soletta's](https://github.com/solettaproject/meta-soletta/tree/master/recipes-modules/low-speed-spidev) - but I've expanded it to work for several devices supplied as an array. We'll see what Ref-OS-IoT folks decide on this one (I've reported my findings to them in https://github.com/01org/ref-os-iot/issues/2 and also internally via email), if my one is going to be useful, I'll publish it on GH.

Another alternative is to use ACPI/DT config, which gets picked up by `spidev` by default (though you need to use specific device IDs, which is not always useful).

commit 32196d10 (2017-06-18): joule: align SPI bus numbering with reality

Also update docs to make the numbering and pins used crystal clear.

Closes #758.
closed 2017-06-18 by alext-mkrs