eclipse-mraa/mraa #759: [edison] support vanilla (> 4.11) kernel

htot, 2017-06-13:
Edison is for a large part supported in vanilla (>4.11) kernels. Moreover a Yocto morty image is available here https://github.com/htot/meta-intel-edison/tree/morty.

Typically, the serial ports are renamed to linux standard naming (/dev/ttyMFD1 replaced by /dev/ttyS1), and pinctrl and pinmux changed. Probably other stuff too.

How would be go about providing support for vanilla kernels? Detect the kernel version from within intel_edison_fab_c.c? Or provide a separate platform abstraction?

arfoll, 2017-06-13:
So I think it makes more sense if the changes are not huge to detect the kernel version in the edison platform init stages and then either change the advance platform functions based on this or change the mapping, depending on how big the changes are.

htot, 2017-06-13:
As I understand with 3.10 pin muxing is controlled from /sys/kernel/debug/gpio_debug/gpioN/current_pinmux and in 4,11, I need to familiarize myself with the standard interface. It's probably not to hard, but I guess we need to enable functions in mraa one by one, as maybe not everything may be supported right now.

htot, 2017-10-04:
Resolved by 6a1ed23

closed 2017-10-04 by htot