eclipse-mraa/mraa #760: How to install in virtualenv?

mgiaco, 2017-06-14:
Hello, 

Can someone help me on how to install mraa in virtualenv? I have already installed the lib on my BBB and it seems to work. But now i would like to install it in a virtualenv for that i activated the virtualenv - rebuild it and run a make install ..
But if i try to run a test app from the virtualenv mraa is not found. So I am doing anything wrong.

thx

arfoll, 2017-06-14:
I think if you activate the virtualenv and then build mraa (make sure to rerun cmake from a clean build dir) it should got to the right place. Check the cmake cache to see the PYTHON2_PACKAGES_PATH and PYTHON3_PACKAGES_PATH are good and override them with your virtualenv packages path if wrong. Another option is just to copy the mraa.py and _mraa.so directly to the packages path?

mgiaco, 2017-06-14:
So I followed your advice. The path was wrong I changed it to the virtualenv path but I does not work. So I have to copy it by hand. By the way mraa also works with python3 - right?

arfoll, 2017-06-14:
so python3 & python2 binaries are different but yes mraa works with both

do you have the output of `cmake`, `cmake -LA` and `VERBOSE=1 make install`? might be able to work something out.

mgiaco, 2017-06-20:
Hey arfoll, i will add this in the next few days for you okay. 
cheers
mathias

arfoll, 2017-06-20:
cool that would be great

mgiaco, 2017-07-02:
Hey @arfoll 

Sorry a bit late but here is what you want:

```
(venv) debian@beaglebone:~/libs/mraa/build$ cmake -LA ..
-- The C compiler identification is GNU 4.9.2
-- The CXX compiler identification is GNU 4.9.2
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Performing Test HAS_C__Wall
-- Performing Test HAS_C__Wall - Success
-- Performing Test HAS_C__Werror_main
-- Performing Test HAS_C__Werror_main - Success
-- Performing Test HAS_C__Wformat
-- Performing Test HAS_C__Wformat - Success
-- Performing Test HAS_C__Wmain
-- Performing Test HAS_C__Wmain - Success
-- Performing Test HAS_C__Wuninitialized
-- Performing Test HAS_C__Wuninitialized - Success
-- Performing Test HAS_C__Winit_self
-- Performing Test HAS_C__Winit_self - Success
-- Performing Test HAS_C__Werror_implicit
-- Performing Test HAS_C__Werror_implicit - Success
-- Performing Test HAS_C__Werror_missing_parameter_type
-- Performing Test HAS_C__Werror_missing_parameter_type - Success
-- Performing Test HAS_CXX__Wall
-- Performing Test HAS_CXX__Wall - Success
-- Performing Test HAS_CXX__Werror_main
-- Performing Test HAS_CXX__Werror_main - Success
-- Performing Test HAS_CXX__Wformat
-- Performing Test HAS_CXX__Wformat - Success
-- Performing Test HAS_CXX__Wmain
-- Performing Test HAS_CXX__Wmain - Success
-- Performing Test HAS_CXX__Wuninitialized
-- Performing Test HAS_CXX__Wuninitialized - Success
-- Performing Test HAS_CXX__Winit_self
-- Performing Test HAS_CXX__Winit_self - Success
-- Performing Test HAS_CXX__Wnon_virtual_dtor
-- Performing Test HAS_CXX__Wnon_virtual_dtor - Success
-- Performing Test HAS_CXX__Woverloaded_virtual
-- Performing Test HAS_CXX__Woverloaded_virtual - Success
-- Performing Test HAS_CXX__Wreorder
-- Performing Test HAS_CXX__Wreorder - Success
-- INFO - libmraa Version v1.7.0-45-gcaf75a6
-- INFO - cmake Version 3.5.2
-- INFO - Target arch is armv7
-- Found PythonInterp: /home/debian/pywork/venv/bin/python2.7 (found suitable version "2.7.9", minimum required is "2.7") 
-- Found PythonLibs: /usr/lib/arm-linux-gnueabihf/libpython2.7.so (found suitable exact version "2.7.9") 
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.4.2", minimum required is "3") 
-- Could NOT find PythonLibs (missing:  PYTHON_LIBRARIES PYTHON_INCLUDE_DIRS) (Required is exact version "3.4.2")
[... 833 lines cut ...]
-- Up-to-date: /usr/local/share/mraa/examples/javascript/GPIO_DigitalRead.js
-- Up-to-date: /usr/local/share/mraa/examples/javascript/uart.js
-- Up-to-date: /usr/local/share/mraa/examples/javascript/GPIO_DigitalWrite.js
-- Up-to-date: /usr/local/share/mraa/examples/mmap-io2.c
-- Up-to-date: /usr/local/share/mraa/examples/firmata_curie_imu.c
-- Up-to-date: /usr/local/share/mraa/examples/i2c_firmata.c
-- Up-to-date: /usr/local/share/mraa/examples/CMakeLists.txt
-- Up-to-date: /usr/local/share/mraa/examples/spi_mcp4261.c
-- Up-to-date: /usr/local/share/mraa/examples/spi_max7219.c
-- Up-to-date: /usr/local/share/mraa/examples/gpio_read6.c
-- Up-to-date: /usr/local/share/mraa/examples/platform
-- Up-to-date: /usr/local/share/mraa/examples/platform/intel-aero.json
-- Up-to-date: /usr/local/share/mraa/examples/platform/turbotjson.json
-- Up-to-date: /usr/local/share/mraa/examples/uart.c
-- Installing: /usr/local/bin/mraa-gpio
-- Set runtime path of "/usr/local/bin/mraa-gpio" to ""
-- Installing: /usr/local/bin/mraa-i2c
-- Set runtime path of "/usr/local/bin/mraa-i2c" to ""
-- Installing: /usr/local/bin/mraa-uart
-- Set runtime path of "/usr/local/bin/mraa-uart" to ""
(venv) debian@beaglebone:~/libs/mraa/build$ 
```
If you need anything more please do not hesitate to contact me.
cheers
mathias

alext-mkrs, 2017-07-03:
Here's the specific reason - it doesn't find the Python 3 libs, thus builds only Python 2 bindings:
```
-- Found PythonInterp: /home/debian/pywork/venv/bin/python2.7 (found suitable version "2.7.9", minimum required is "2.7") 
-- Found PythonLibs: /usr/lib/arm-linux-gnueabihf/libpython2.7.so (found suitable exact version "2.7.9") 
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.4.2", minimum required is "3") 
>>>>>> -- Could NOT find PythonLibs (missing:  PYTHON_LIBRARIES PYTHON_INCLUDE_DIRS) (Required is exact version "3.4.2")
```
You do have them installed, don't you?

mgiaco, 2017-07-03:
Yes I only use python2.7 now but I will switch to python3. Is it a good idea to install booth on the same image. Do I really need the exact version 3.4.2?

arfoll, 2017-07-04:
There "should" be no issue using any python3 to build. But we don't test that scenario so there may be a few weird ones. Having both on the same image is really no issue, pretty much all distros do this.

Btw you can build mraa in a recipe too, or you can build it with your OE toolchain with our `cmake/Toolchains/oe-sdk_cross.cmake`. See the docs/building.md for more details

mgiaco, 2017-07-05:
Okay I will try that, many thanks.

closed 2017-07-05 by mgiaco