eclipse-mraa/mraa #761: Platform loading Segmentation Fault

nigeltiany, 2017-06-15:
I am trying to platform load the Odroid XU4. 
Here is my odroid.json
```json
{
    "version": "0.0.1",
    "platform": [
        {
            "id": 1010,
            "name": "Odroid",
            "pin_count": 32, // 30 GPIO Plus 2 Analog inputs
            "gpio_count": 30,
            "aio_count": 2,
            "uart_count":1,
            "i2c_count": 4,
            "spi_count": 1,
            "pwmDefPeriod": 0,
            "pwmMaxPeriod": 0,
            "pwmMinPeriod": 0
        }
    ],
    "layout": [
        {"label": "AIN-0"},
        {"label": "GPIO-4"},
        {"label": "GPIO-5"},
        {"label": "GPIO-6"},
        {"label": "GPIO-7"},
        {"label": "GPIO-8"},
        {"label": "GPIO-9"},
        {"label": "GPIO-10"},
        {"label": "GPIO-11"},
        {"label": "GPIO-13"},
        {"label": "GPIO-14"},
        {"label": "GPIO-15"},
        {"label": "GPIO-16"},
        {"label": "GPIO-17"},
        {"label": "GPIO-18"},
        {"label": "GPIO-19"},
        {"label": "GPIO-20"},
        {"label": "GPIO-21"},
        {"label": "GPIO-22"},
        {"label": "AIN-3"},
        {"label": "GPIO-24"},
        {"label": "GPIO-25"},
        {"label": "GPIO-26"},
        {"label": "GPIO-27"},
        {"label": "X-GPIO-4"},
        {"label": "X-GPIO-5"},
        {"label": "X-GPIO-6"},
        {"label": "X-GPIO-7"},
        {"label": "X-GPIO-9"},
        {"label": "X-GPIO-10"},
        {"label": "X-GPIO-11"},
        {"label": "X-GPIO-12"}
    ],
    "AIO": [
        { "index": 0, "rawpin": "in_voltage0_raw"},
        { "index": 19, "rawpin": "in_voltage3_raw"}
    ],
    "GPIO": [
        { "index": 1, "rawpin": 173 },
        { "index": 2, "rawpin": 174 },
        { "index": 3, "rawpin": 171 },
        { "index": 4, "rawpin": 192 },
        { "index": 5, "rawpin": 172 },
        { "index": 6, "rawpin": 191 },
        { "index": 7, "rawpin": 189 },
        { "index": 8, "rawpin": 190 },
        { "index": 9, "rawpin": 21 },
        { "index": 10, "rawpin": 210 },
        { "index": 11, "rawpin": 18 },
        { "index": 12, "rawpin": 209 },
        { "index": 13, "rawpin": 22 },
        { "index": 14, "rawpin": 19 },
        { "index": 15, "rawpin": 30 },
        { "index": 16, "rawpin": 28 },
        { "index": 17, "rawpin": 29 },
        { "index": 18, "rawpin": 31 },
        { "index": 20, "rawpin": 25 },
        { "index": 21, "rawpin": 23 },
        { "index": 22, "rawpin": 24 },
        { "index": 23, "rawpin": 33 },
        //Twelve Pin Header
        { "index": 24, "rawpin": 187 },
        { "index": 25, "rawpin": 34 },
        { "index": 26, "rawpin": 188 },
        { "index": 27, "rawpin": 225 },
        { "index": 28, "rawpin": 226 },
        { "index": 29, "rawpin": 229 },
        { "index": 30, "rawpin": 227 },
        { "index": 31, "rawpin": 228 }
    ],
    "SPI": [
        {   
            "index": 0, 
            "chipID": 1010, 
            "clock": 7, 
            "miso": 6,
            "mosi": 4,
            "chipselect": 8 
        } 
    ],
    "I2C": [
        { 
            "index": 0,
            "rawpin": 210,
            "sclpin": 10,
            "default": false
        },
        { 
            "index": 1,
            "rawpin": 209,
            "sdapin": 12,
            "default": false
        },
        {
            "index": 2,
            "rawpin": 188,
            "sclpin": 26,
            "default": true
        },
        {
            "index": 3,
            "rawpin": 187,
            "sdapin": 24,
            "default": true
        }
    ],
    "UART": [
        {
            "index": 0,
            "chipID": 1010,
            "rawpin": 171,
            "rx": 3,
            "path": "/dev/ttySAC0",
            "default": false
        },{
            "index": 1,
            "chipID": 1010,
            "rawpin": 172,
            "tx": 5,
            "path": "/dev/ttySAC0",
            "default": false
        }
    ]
}
```

Here is my javascript file

```javascript
"use strict";

var mraa = require('mraa-odroid');
mraa.initJsonPlatform('./odroid.json');

console.log('MRAA Version: ' + mraa.getVersion()); //write the mraa version to the console
console.log(new mraa.Gpio(3,false));
```
```shell $node theAboveJavascript.js``` returns 
```shell 
odroid@odroid:~/Documents/mraa-odroid-test$ sudo node index
MRAA Version: v1.7.0
Segmentation fault
```

Also after placing the json file in ```etc/odroid-mraa.json``` and running
```shell $export MRAA_JSON_PLATFORM="/etc/odroid-mraa.json"```
Then running ```shell $ mraa-gpio list``` i get ```No Pins```  as Out Put

alext-mkrs, 2017-06-15:
I think you may at least be hitting #758 as some of your pin labels are longer than 8 symbols. I'm getting really close to fixing that (in terms of time, technically the fix is simple), but in the meanwhile try shortening your pin labels.

alext-mkrs, 2017-06-15:
Actually, I've went ahead and created the fix for that one, see PR #763. It may not be the only (or at all) thing you're hitting here, but at least I think it plays a role.

alext-mkrs, 2017-06-15:
Another thing I'd be suspicious about are those `//` comments, AFAIU that's not valid JSON, so you may want to remove them as well.

nigeltiany, 2017-06-15:
Merged the PR in my fork and build the source.
I have removed SPI,UART and I2C configurations and the comments (configuring the json on SPI,UART and I2C is a bit unnatural but anyway). The json now only has the gpio pins.
With that ``` $ mraa-gpio list``` lists the pins successfully even when the label character are more than 11. 

The js code still throws the segmentation fault error unless i run the code as a super user. This was not the case before. Running as the super user still yielded segmentation error

With out super user
```
MRAA Version: v1.7.0-36-g28b03cf
Segmentation fault
```

As super user
```
MRAA Version: v1.7.0-36-g28b03cf
Gpio {}
*** Error in `node': double free or corruption (fasttop): 0x01356d90 ***
Aborted
```

alext-mkrs, 2017-06-17:
Ok, thanks for reporting back and let's dissect this as it seems like you're now talking about something else.

To make sure - now that you have my fix compiled in (+removed comments and bus configurations), the code doesn't fail the same way it did, right? I take your comment about `mraa-gpio list` as a confirmation of that, but let's double-check.

Now, with the fix in, you still see a failure when running your code. I have several questions to understand the situation better:
1. You `require mraa-odroid`, which is not the standard name for mraa's JS binding. What did you change to alter that?
1. Mraa requires root permissions to run properly, so there's no point in running under limited accounts.
1. Could you please post your current version of the JSON file, so that I could try it on my machine?

nigeltiany, 2017-06-17:
I renamed the contents of the built javascript folder to mraa-odroid. 
Here is the current json file.

```
{
    "version": "0.0.1",
    "platform": [
        {
            "id": 1010,
            "name": "Odroid",
            "pin_count": 30,
            "gpio_count": 30,
            "aio_count": 0,
            "uart_count":0,
            "i2c_count": 0,
            "spi_count": 0,
            "pwmDefPeriod": 0,
            "pwmMaxPeriod": 0,
            "pwmMinPeriod": 0
        }
    ],
    "layout": [
        {"label": "GPIO-4"},
        {"label": "GPIO-5"},
        {"label": "GPIO-6"},
        {"label": "GPIO-7"},
        {"label": "GPIO-8"},
        {"label": "GPIO-9"},
        {"label": "GPIO-10"},
        {"label": "GPIO-11"},
        {"label": "GPIO-13"},
        {"label": "GPIO-14"},
        {"label": "GPIO-15"},
        {"label": "GPIO-16"},
        {"label": "GPIO-17"},
        {"label": "GPIO-18"},
        {"label": "GPIO-19"},
        {"label": "GPIO-20"},
        {"label": "GPIO-21"},
        {"label": "GPIO-22"},
        {"label": "GPIO-24"},
        {"label": "GPIO-25"},
        {"label": "GPIO-26"},
        {"label": "GPIO-27"},
        {"label": "X-GPIO-4"},
        {"label": "X-GPIO-5"},
        {"label": "X-GPIO-6"},
        {"label": "X-GPIO-7"},
        {"label": "X-GPIO-9"},
        {"label": "X-GPIO-10"},
        {"label": "X-GPIO-11"},
        {"label": "X-GPIO-12"}
    ],
    "GPIO": [
        { "index": 0, "rawpin": 173 },
        { "index": 1, "rawpin": 174 },
        { "index": 2, "rawpin": 171 },
        { "index": 3, "rawpin": 192 },
        { "index": 4, "rawpin": 172 },
        { "index": 5, "rawpin": 191 },
        { "index": 6, "rawpin": 189 },
        { "index": 7, "rawpin": 190 },
        { "index": 8, "rawpin": 21 },
        { "index": 9, "rawpin": 210 },
        { "index": 10, "rawpin": 18 },
        { "index": 11, "rawpin": 209 },
        { "index": 12, "rawpin": 22 },
        { "index": 13, "rawpin": 19 },
        { "index": 14, "rawpin": 30 },
        { "index": 15, "rawpin": 28 },
        { "index": 16, "rawpin": 29 },
        { "index": 17, "rawpin": 31 },
        { "index": 18, "rawpin": 25 },
        { "index": 19, "rawpin": 23 },
        { "index": 20, "rawpin": 24 },
        { "index": 21, "rawpin": 33 },
        { "index": 22, "rawpin": 187 },
        { "index": 23, "rawpin": 34 },
        { "index": 24, "rawpin": 188 },
        { "index": 25, "rawpin": 225 },
        { "index": 26, "rawpin": 226 },
        { "index": 27, "rawpin": 229 },
        { "index": 28, "rawpin": 227 },
        { "index": 29, "rawpin": 228 }
    ]
}
```

alext-mkrs, 2017-06-27:
Not yet, sorry. I was planning to look into this during the weekend, but then something else came in and pre-empted that. I'll try to find some time during the week or on the upcoming weekend.

What could additionally help me is a syslog dump around the time you run your code. There could be some error/diag messages from the library.

arfoll, 2017-06-28:
This isn't very useful but I can load the platform file and it seems to work:
```
130 brendan@schultz ~/git/mraa/build/src/python/python3 (git)-[master] % python3                                                                                            :(
Python 3.6.1 (default, Mar 27 2017, 00:27:06) 
[GCC 6.3.1 20170306] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import mraa as m
>>> m.initJsonPlatform("/home/brendan/git/mraa/odroid.json")
0
>>> m.getPlatformName()
'Odroid'
```

Obviously just doing it on my PC so it means very little. Could really do with a journalctl/syslog output...

nigeltiany, 2017-06-29:
Here it is
```
Jun 29 02:13:32 odroid sudo[6097]:   odroid : TTY=pts/1 ; PWD=/home/odroid/Documents/
Jun 29 02:13:32 odroid sudo[6097]: pam_unix(sudo:session): session opened for user ro
Jun 29 02:13:32 odroid libmraa[6101]: libmraa version v1.7.0-36-g28b03cf initialised 
Jun 29 02:13:32 odroid libmraa[6101]: Unknown Platform, currently not supported by MR
Jun 29 02:13:32 odroid libmraa[6101]: libmraa initialised for platform 'Unknown platf
Jun 29 02:13:32 odroid libmraa[6101]: init_json_platform: AIO wasn't found in the jso
Jun 29 02:13:32 odroid libmraa[6101]: init_json_platform: SPI wasn't found in the jso
Jun 29 02:13:32 odroid libmraa[6101]: init_json_platform: I2C wasn't found in the jso
Jun 29 02:13:32 odroid libmraa[6101]: init_json_platform: UART wasn't found in the js
Jun 29 02:13:32 odroid libmraa[6101]: init_json_platform: PWM wasn't found in the jso
Jun 29 02:13:32 odroid libmraa[6101]: init_json_platform: Platform Odroid initialised
Jun 29 02:13:32 odroid sudo[6097]: pam_unix(sudo:session): session closed for user ro
Jun 29 02:15:19 odroid systemd[1]: Starting Cleanup of Temporary Directories...
Jun 29 02:15:19 odroid systemd-tmpfiles[6162]: [/usr/lib/tmpfiles.d/var.conf:14] Dupl
Jun 29 02:15:19 odroid systemd[1]: Started Cleanup of Temporary Directories.
Jun 29 02:16:16 odroid ntpd[1182]: 78.46.93.106 local addr 192.168.1.80 -> <null>
Jun 29 02:16:22 odroid ntpd[1182]: 149.202.200.140 local addr 192.168.1.80 -> <null>
Jun 29 02:17:01 odroid CRON[6217]: pam_unix(cron:session): session opened for user ro
Jun 29 02:17:01 odroid CRON[6221]: (root) CMD (   cd / && run-parts --report /etc/cro
Jun 29 02:17:01 odroid CRON[6217]: pam_unix(cron:session): session closed for user ro
lines 14-41/41 (END)

```

nigeltiany, 2017-06-29:
In python after running ```$ export MRAA_JSON_PLATFORM="/etc/odroid-mraa.json"``` again
```python
>>> import mraa as m
>>> m.initJsonPlatform("./odroid.json")
0
>>> m.getPlatformName()
'Odroid'
>>> x = m.Gpio(0)

Traceback (most recent call last):
  File "<pyshell#3>", line 1, in <module>
    x = m.Gpio(0)
  File "/usr/lib/python2.7/dist-packages/mraa.py", line 1063, in __init__
    this = _mraa.new_Gpio(pin, owner, raw)
ValueError: Invalid GPIO pin specified
>>> x = m.Gpio(23)

Traceback (most recent call last):
  File "<pyshell#4>", line 1, in <module>
    x = m.Gpio(23)
  File "/usr/lib/python2.7/dist-packages/mraa.py", line 1063, in __init__
    this = _mraa.new_Gpio(pin, owner, raw)
ValueError: Invalid GPIO pin specified
>>> x = m .gpioFromDesc("GPIO-4")

Traceback (most recent call last):
  File "<pyshell#6>", line 1, in <module>
    x = m.gpioFromDecsc("GPIO-4")
AttributeError: 'module' object has no attribute 'gpioFromDecsc'
>>> x = m.Gpio(4)

Traceback (most recent call last):
  File "<pyshell#7>", line 1, in <module>
    x = m.Gpio(4)
  File "/usr/lib/python2.7/dist-packages/mraa.py", line 1063, in __init__
    this = _mraa.new_Gpio(pin, owner, raw)
ValueError: Invalid GPIO pin specified
>>> 
```

```
Jun 29 02:32:05 odroid libmraa[10300]: init_json_platform: UART wasn't found in the json file
Jun 29 02:32:05 odroid libmraa[10300]: init_json_platform: PWM wasn't found in the json file
Jun 29 02:32:05 odroid libmraa[10300]: init_json_platform: Platform Odroid initialised via json
Jun 29 02:33:18 odroid libmraa[10300]: gpio173: init: Failed to open 'export' for writing: Permission denied
Jun 29 02:34:29 odroid libmraa[10388]: libmraa version v1.7.0-36-g28b03cf initialised by user 'odroid' with EUID 1000
Jun 29 02:34:29 odroid libmraa[10388]: Unknown Platform, currently not supported by MRAA
Jun 29 02:34:29 odroid libmraa[10388]: libmraa initialised for platform 'Unknown platform' of type 98
Jun 29 02:35:06 odroid libmraa[10300]: gpio34: init: Failed to open 'export' for writing: Permission denied
Jun 29 02:37:09 odroid libmraa[10300]: gpio172: init: Failed to open 'export' for writing: Permission denied
Jun 29 02:40:07 odroid libmraa[10447]: libmraa version v1.7.0-36-g28b03cf initialised by user 'odroid' with EUID 1000
Jun 29 02:40:07 odroid libmraa[10447]: init_json_platform: AIO wasn't found in the json file
Jun 29 02:40:07 odroid libmraa[10447]: init_json_platform: SPI wasn't found in the json file
Jun 29 02:40:07 odroid libmraa[10447]: init_json_platform: I2C wasn't found in the json file
Jun 29 02:40:07 odroid libmraa[10447]: init_json_platform: UART wasn't found in the json file
Jun 29 02:40:07 odroid libmraa[10447]: init_json_platform: PWM wasn't found in the json file
Jun 29 02:40:07 odroid libmraa[10447]: init_json_platform: Platform Odroid initialised via json
Jun 29 02:40:07 odroid libmraa[10447]: libmraa initialised for platform 'Odroid' of type 97
Jun 29 02:40:43 odroid libmraa[10300]: gpio172: init: Failed to open 'export' for writing: Permission denied
```

Then with nodejs one more time

```
Jun 29 02:44:24 odroid sudo[10492]:   odroid : TTY=pts/1 ; PWD=/home/odroid/Documents/drone-v2 ; USER=root ; COMMAND=/usr/bin/node index
Jun 29 02:44:24 odroid sudo[10492]: pam_unix(sudo:session): session opened for user root by odroid(uid=0)
Jun 29 02:44:24 odroid libmraa[10496]: libmraa version v1.7.0-36-g28b03cf initialised by user 'root' with EUID 0
Jun 29 02:44:24 odroid libmraa[10496]: Unknown Platform, currently not supported by MRAA
Jun 29 02:44:24 odroid libmraa[10496]: libmraa initialised for platform 'Unknown platform' of type 98
Jun 29 02:44:24 odroid libmraa[10496]: init_json_platform: AIO wasn't found in the json file
Jun 29 02:44:24 odroid libmraa[10496]: init_json_platform: SPI wasn't found in the json file
Jun 29 02:44:24 odroid libmraa[10496]: init_json_platform: I2C wasn't found in the json file
Jun 29 02:44:24 odroid libmraa[10496]: init_json_platform: UART wasn't found in the json file
Jun 29 02:44:24 odroid libmraa[10496]: init_json_platform: PWM wasn't found in the json file
Jun 29 02:44:24 odroid libmraa[10496]: init_json_platform: Platform Odroid initialised via json
Jun 29 02:44:24 odroid sudo[10492]: pam_unix(sudo:session): session closed for user root

```

alext-mkrs, 2017-07-01:
Thanks for log snippets. The last JS example you posted fails due to the fact you don't initialize the platform in it - you just import mraa and then try to instantiate a GPIO 0 object, but mraa doesn't know what platform this is. However in your example in the first post you did the needful and that ended in segfault. Interesting that there are no traces of that in the syslog. And the fact that Python works and JS doesn't is also useful, let me look into this in more detail.

alext-mkrs, 2017-07-01:
At the end of the day, this is a bug caused by incorrect string/pointer manipulations, leading to a double free. It was actually happening in Python as well, but it seems to swallow this a bit better than Node.js, so it was not apparent.

The fix is coming in #779.

commit caf75a68 (2017-07-01): jsonplatform.c: properly reallocate memory for platform name

Fixes #761.
closed 2017-07-01 by arfoll