eclipse-mraa/mraa #764: Intel Joule duplicates GPIO_22

jewalt, 2017-06-16:
MRAA pin 1 used to be BTN_N on the schematic (446). Someone replaced it with GPIO_22 which is already in the pin map at 84.  Why duplicate it and break old functionality?

Propanu, 2017-06-16:
@jewalt, I made the change. There are 80 pins on the expansion board, MRAA doesn't map past pin 80 except the 4 LEDs (100+). GPIO_22 is actually exposed on both the expansion and the DFRobot shield, and labeled accordingly. However, I wouldn't mind exposing 446 as pin 81 if you find it useful for your application.

jewalt, 2017-06-16:
I don't understand why pin 1 was remapped?  It breaks backward compatibility (mraa 1.5) and mapping GPIO_22 twice is useless.  The MRAA pin numbers are arbitrary so why the remap?  Mapping 446 to 81 would be useful but does not seem like the right way to solve this issue.

Propanu, 2017-06-17:
I disagree, MRAA pin numbers are not arbitrary at all. We've updated the [pin mapping](../blob/master/docs/joule.md) to match the J12 and J13 headers. GPIO_22 (451) was not in the board file before, but you could still access it with `mraa_gpio_init_raw(...)`, just like you can BTN_N now with (446). Is this what you mean by mapped twice?

jewalt, 2017-06-17:
GPIO_22 (451) is at 1 (now) and 84.  I assume 84 is a carry-over from old code and I'm glad it wasn't removed.  I would like to see BTN_N (446) added it will be used on many custom board designs.  

I was unaware of the use of the *raw functions.  Thank you for the pointer.  I think that will solve all of my problems.

closed 2017-06-17 by jewalt

alext-mkrs, 2017-06-18:
While raw functions seem to solve your original problem, I'm curious too, how did you manage to find it at index 84, if it was never there? As you can [see](https://github.com/intel-iot-devkit/mraa/blame/master/src/x86/intel_joule_expansion.c#L673-L676), that part of pin indexes, 81-99, is filled with stub and inactive pin definitions, so they couldn't be used. And that part of the code stems from the original commit adding Joule support, so it's really unclear how come you can see that GPIO_22/451 pin at that index. If you could elaborate on that, we'd be glad to fix or clarify the duplication.

jewalt, 2017-07-03:
Sorry it took so long to get back to this, I got side-tracked on another project.  In early November, I submitted a patch that fixed quite a few GPIO names for the tuchuck board as it was a mess at the time.  In that patch, I added GPIO_22 at 84.  I see now that it should have been at 1 instead but that wasn't the reason the patch was rejected.  Since my perfectly workable patch was rejected, I decided not to submit further patches and simply work on my own.   So all of my work that uses MRAA was using my version that had proper names for other pins.  I see that parts of my patch were taken and used but not all so when I pulled down a new version of MRAA, lots of things broke.