eclipse-mraa/mraa #766: how to install then on the target?

mgiaco, 2017-06-19:
Hi, I followed the guide for building mraa via yocto sdk. It seems to work but after the build what is the right way to get those build on the target - sorry for the maybe silly question.

mgiaco, 2017-06-20:
So no one uses yocto?
I also tried the intel iot recipes but i looks unmaintained for me?

arfoll, 2017-06-20:
Many do it, but our recipes are a bit of an unmaintained mess all over the place, we don't have a good central place. The best I know of currently is https://github.com/intel/intel-iot-refkit/tree/master/meta-refkit-core/recipes-sensors.

mgiaco, 2017-06-20:
Okay many thanks I will try that - but if i compile it via the cross sdk - how can i install it on the target then?  
So okay I can copy and paste the *.so and py files at the right place. But is there a way to install them after build on the cross sdk?

arfoll, 2017-06-20:
I'm not sure which cross SDK you are referring to. If you create your own packages for your device you really have to create a cross SDK for your device too. That's the "proper" way of doing things. Obviously copying binaries works too :)

alext-mkrs, 2017-06-20:
I guess @mgiaco is asking about a scenario where mraa is being built on the host by using cross SDK generated by Yocto, is that correct? If so, that's a bit unnatural way of doing that, but generally yes, you need to manually copy generated files into proper places, there's no "normal" of installing files built like that, like "make install" - as this way of building doesn't have the means of installing built binaries.

The files are, OTMH: main .so, python, nodejs, java bindings (whichever of those you build). Specific locations depend on details of your target.

mgiaco, 2017-06-21:
@alext-mkrs  Hey - yes that was my question. No I have integrated all the build tools in the yocto image and build mraa on the device but for production I would like to have a small image. So I hope I will find a working recipe.

Or what is the natural way?

alext-mkrs, 2017-06-22:
I see. So having Yocto and embedded system in the context, there are three basic ways of building software for your target (board):
1. Build on host, using Yocto itself (not the SDK). This is typically used for building OS images or additional SW for it. SW typically gets packaged to something used by the OS image (ipk, rpm, deb). This depends on the settings, but typically some form of package management is enabled everywhere. This would be the way I'd recommend you to follow to get mraa built.
1. Build on host, using the SDK, generated by Yocto for the OS image you've built. This is generally a way for building your "user" or "functional" applications you want to run on top of your target's OS+library set. Building mraa (or other libraries, to that matter) may be more difficult at least because of the fact there's no packaging when using this one. Then, there's e.g. no direct support for CMake-based projects, unlike in the Yocto way above. So technically you can do that, practically I'd not recommend it for mraa building.
1. Building on the target. You tried that already. Simplest of all three, but can be slow (depending on the target) and may require target configuration (installing additional headers, libraries).

All in all, I'd recommend you to either build on target, or use the Yocto mraa recipe @arfoll referred to in your other issue and build via Yocto. Using SDK would be my last resort of all three.

mgiaco, 2017-06-22:
Okay many thanks I will try that.

closed 2017-06-22 by mgiaco