eclipse-mraa/mraa #767: Building mraa on yocto

Respekt1, 2017-06-19:
Hello Community,

I 'am trying to build mraa for a custom yocto build. The processor is the same as beaglebone black. On 
target and crosscompilation I get the error:

```bash
/opt/poky/2.1.3/mraa/build⟫ cmake -DCMAKE_TOOLCHAIN_FILE=../cmake/Toolchains/oe-sdk_cross.cmake ..             
-- INFO - libmraa Version v1.7.0-40-ge0052ac
-- INFO - cmake Version 3.5.1
-- INFO - Target arch is armv7
-- INFO - Adding onewire backend support
-- INFO - Adding support for platform ALL
-- INFO - Adding support for all platforms
INFO - Adding ARM platforms
CMake Warning at src/CMakeLists.txt:186 (message):
  Enabled JSONPLAT but JSON-C library not found: Ignoring JSONPLAT

PYTHON2 attempting to build!
ERROR - node.h not found
ERROR - v8.h not found
ERROR - uv.h not found
CMake Error at /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find Nodejs (missing: NODEJS_INCLUDE_DIRS)
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:388 (_FPHSA_FAILURE_MESSAGE)
  cmake/modules/FindNodejs.cmake:55 (find_package_handle_standard_args)
  src/javascript/CMakeLists.txt:1 (find_package)

-- Configuring incomplete, errors occurred!
See also "/opt/poky/2.1.3/mraa/build/CMakeFiles/CMakeOutput.log".
See also "/opt/poky/2.1.3/mraa/build/CMakeFiles/CMakeError.log".
```

How can I solve the NODEJS_INCLUDE_DIRS? 

Thanks for your help.

mgiaco, 2017-06-20:
@Respekt1  take a look at my issue - maybe that is also a solution for you.

alext-mkrs, 2017-06-20:
@Respekt1, how does your mraa recipe file look like? Generally, you have to either declare dependency on "nodejs" or manually specify things like `NODEJS_INCLUDE_DIRS`, `NODEJS_EXECUTABLE`, `NODEJS_ROOT_DIR` in the cmake parameters in the recipe.

Or just remove the Node.js binding from the build process, but that's sort of obvious and I assume you actually need it.

mgiaco, 2017-07-24:
@Respekt1 did you check the hint from @alext-mkrs ?

alext-mkrs, 2017-08-10:
@Respekt1, is this still actual?

Respekt1, 2017-08-21:
@alext-mkrs: I build mraa with this commands:

```bash
cd ~
git clone https://github.com/intel-iot-devkit/mraa.git && cd mraa && \
mkdir build && cd build && \
cmake -DBUILDSWIGNODE=OFF -DCMAKE_INSTALL_PREFIX:H=/usr .. && \
make install 
```

thanks for support

closed 2017-08-21 by Respekt1

alext-mkrs, 2017-08-24:
Did you really want to close the issue or just clicked the wrong button?

So you are actually building using a cross-compiler from the Yocto-generated SDK, do I get it right? Or you're building on the target?

Generally, whether it's a recipe or cross-compilation, you have to set those defines I mentioned earlier. If you're building in the target, you need to make sure you've installed the Node development package, so that headers are available.

Finally, you shouldn't really be having that problem if you set `BUILDSWIGNODE=OFF`. I don't fully understand your setup as you seem to try different things in your posts, so let's settle on one case or just elaborate on all of them.