eclipse-mraa/mraa #771: Latests commits break openaps on edison with ValueError: Invalid GPIO pin specified error

PieterGit, 2017-06-23:
Multiple users of openaps reported that the latest master checkout causes
`ValueError: Invalid GPIO pin specified` with openaps. Previous master checkouts didn't have that error.

Latest comit that was ok:
https://github.com/intel-iot-devkit/mraa/commit/e0052acfa7ab441b5417a0fbd010ce77b285a4d8

So the problem was caused by the following commits:
- https://github.com/intel-iot-devkit/mraa/commit/30bbb886850d2a7c55c5803e5c830e5c6cfd2e1a
- https://github.com/intel-iot-devkit/mraa/commit/d544e3c2f60ff3dd1b2501fb6e8b4136347675f3

Stavrosnco, 2017-06-23:
I was able to reproduce this issue after performing installation of the current master branch of oref0 with openaps.

The command producing the error was:
`openaps mmtune`

Which produced this stack trace:

`Traceback (most recent call last):
  File "/usr/local/bin/openaps-report", line 82, in <module>
    app( )
  File "/usr/local/lib/python2.7/dist-packages/openaps/cli/__init__.py", line 51, in __call__
    self.run(self.args)
  File "/usr/local/bin/openaps-report", line 75, in run
    output = app(args, self)
  File "/usr/local/lib/python2.7/dist-packages/openaps/cli/subcommand.py", line 52, in __call__
    return self.method.main(args, app)
  File "/usr/local/lib/python2.7/dist-packages/openaps/reports/invoke.py", line 40, in main
    output = task.method(args, app)
  File "/usr/local/lib/python2.7/dist-packages/openaps/uses/use.py", line 44, in __call__
    self.before_main(args, app)
  File "/usr/local/lib/python2.7/dist-packages/openaps/vendors/medtronic.py", line 62, in before_main
    self.setup_medtronic( )
  File "/usr/local/lib/python2.7/dist-packages/mmeowlink/vendors/mmeowlink.py", line 92, in setup_medtronic
    setup_medtronic_link(self)
  File "/usr/local/lib/python2.7/dist-packages/mmeowlink/vendors/mmeowlink.py", line 75, in setup_medtronic_link
    link = builder.build(radio_type, port)
  File "/usr/local/lib/python2.7/dist-packages/mmeowlink/link_builder.py", line 16, in build
    return SubgRfspyLink(port)
  File "/usr/local/lib/python2.7/dist-packages/mmeowlink/vendors/subg_rfspy_link.py", line 55, in __init__
    self.open()
  File "/usr/local/lib/python2.7/dist-packages/mmeowlink/vendors/serial_interface.py", line 27, in open
    self.serial = spi_serial.SpiSerial()
  File "/usr/local/lib/python2.7/dist-packages/spi_serial/spi_serial.py", line 7, in __init__
    self.cs0 = m.Gpio(23)
  File "/usr/local/lib/python2.7/dist-packages/mraa.py", line 649, in __init__
    this = _mraa.new_Gpio(*args)
ValueError: Invalid GPIO pin specified`

This error seems to occur on any function call involving the radio on the explorer board.

The error also occurs when calling reset_spi_serial:
`Resetting spi_serial
Traceback (most recent call last):
  File "/usr/local/bin/reset_spi_serial.py", line 3, in <module>
    s = spi_serial.SpiSerial()
  File "/usr/local/lib/python2.7/dist-packages/spi_serial/spi_serial.py", line 7, in __init__
    self.cs0 = m.Gpio(23)
  File "/usr/local/lib/python2.7/dist-packages/mraa.py", line 649, in __init__
    this = _mraa.new_Gpio(*args)
ValueError: Invalid GPIO pin specified`

alext-mkrs, 2017-06-24:
I'm looking into this.

alext-mkrs, 2017-06-24:
In the meanwhile, it would help me if you could post the contents of your syslog log files around the time you get the error. There should be additional output from mraa, which could help me understand this better.

arfoll, 2017-06-24:
my guess is that the strncmp in x86.c broke it, sometimes reading from sysfs is not what you expect and we may have had some trailing characters we previously ignored. But syslog output would confirm it

PieterGit, 2017-06-24:
@alext-mkrs thanks for looking into this. I created a hotfix for our OpenAPS community to use the latest stable 1.7.0 MRAA version. We shouldn't have releases which depend on GitHub master/dev branches. I only created the fix, and don't have a spare Edison to help you debug this issue at this moment. I'm pretty sure somebody from https://gitter.im/nightscout/intend-to-bolus?at=594e26d26ae41d5d337c3700 will respond soon.

alext-mkrs, 2017-06-24:
Yeah, no worries, I've found it already - indeed we get the newline symbol attached to the name by `getline()` we use in `x86.c`. Looks like we just need to do some more sanitization on the data we get from there. I'm on it.

alext-mkrs, 2017-06-24:
The fix is coming in PR #772.

PieterGit, 2017-06-24:
Thanks a lot.  Will there be a new MRAA release (with this PR off course) in the coming weeks? At openaps/oref0 we're about to ship a 0.5.0 release and I wonder if we can then ship a 1.7.1 or 1.8.0 MRAA release. I don't know if there are advantages for us, but I noticed the 1.7.0 release is from May.

alext-mkrs, 2017-06-24:
I'd let @arfoll comment on the release plans, but generally I don't see anything obviously beneficial for OpenAPS, given it's using Edison and the feature set used is stable. Since 1.7.0 there were mainly fixes for static code analysis findings (none were critical though) and some ARM-based platforms and Joule cleanups.

arfoll, 2017-06-24:
@PieterGit I don't want to rush a release out without testing, there is not much of interest for you for an edison platform anyways as @alext-mkrs mentioned. I'll see if we can't make a release in the coming weeks.

For your use case you should definately be on a stable version of mraa, I'm a little surprised you are using master in the first place, although always happy to get the bugs 👍 and generally mraa master is pretty stable.

closed 2017-06-24 by arfoll

PieterGit, 2017-06-24:
Please don't hurry with a release 😄 
Most of the OpenAPS users use an Explorer board with Edison. Some still use RPI2/RPI3 (arm based). It was a bug on the oref0 side that master was used instead of a release. We fixed that now with a hotfix / 0.4.3 release.  I' created a PR so that we can confirm that your master branch works with oref0 dev, see
https://github.com/openaps/oref0/pull/507

Can you indicate if the ARM-based cleanups are interesting for our OpenAPS users?

alext-mkrs, 2017-06-24:
You can see full list of commits [here](https://github.com/intel-iot-devkit/mraa/compare/v1.7.0...master), but as for ARM, there was a [RPi fix](https://github.com/intel-iot-devkit/mraa/commit/ecb53c8501439da95e572e54fa138343ae843e96) for setting SPI frequency and an [update for BBB](https://github.com/intel-iot-devkit/mraa/commit/ff03b2de1d8cb1298d6aff7d82bab5362ff87578) to work with newer kernels.

commit 3f932ac9 (2017-06-24): x86.c: sanitize data read from sysfs during board detection

Fixes #771.