eclipse-mraa/mraa #775: Raspberry Pi Issues

ncrastanaren, 2017-06-28:
Hi guys, me again!

Trying to get mraa up and running with a Pi. I've got a Pi 2 running raspbian `Linux raspberrypi 4.9.28-v7+ #998 SMP Mon May 15 16:55:39 BST 2017 armv7l GNU/Linux`. 

Basic GPIO stuff seems to work, but I'm getting nothing out of the I2C lines or SPI lines, according to the logic analyzer. Running `mraa-i2c list` does show an I2C bus, but running `mraa-i2c detect 0` doesn't work, the `i2c` struct is NULL after calling `mraa_i2c_init`. 

Has mraa been tested with newer kernels on the Pi? Is the Pi 2 too old? I fear that I may be running into similar issues as the Beaglebone a few weeks ago.

-Nick

arfoll, 2017-06-28:
Possible but it could be just that the kernel module for spi is not loaded and for i2c could just be bus enumeration on Linux - we have moved to using `mraa_find_i2c_bus_pci` which I would recommend as it's more reliable.

As for testing, you're right rpi was tested on much older kernels - what RPI version are you on?

ncrastanaren, 2017-06-28:
I'm on The Pi 2 running the June 2017 version of Raspbian (kernel v4.9).

ncrastanaren, 2017-06-28:
@arfoll turns out the SPI and I2C modules weren't loaded. I didn't realize that the Pi doesn't load them by default. SPI and I2C seem to work like a charm now.

-Nick

closed 2017-06-28 by ncrastanaren