eclipse-mraa/mraa #777: NPM install failing on Ubuntu Core installed on Dragonboard 410c

Gruutak, 2017-06-30:
Hi, I'm trying to run npm install mraa on my Dragonboard 410c and it's failing.
Ubuntu Core version: 16.04

Log:

`../src/mraa.c: In function ‘imraa_init’:
../src/mraa.c:154:2: error: #error mraa_ARCH NOTHING
 #error mraa_ARCH NOTHING
  ^
mraa.target.mk:136: recipe for target 'Release/obj.target/mraa/src/mraa.o' failed
make: *** [Release/obj.target/mraa/src/mraa.o] Error 1
make: Leaving directory '/home/gruutak/TARCplusplus-iot/node_modules/mraa/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Linux 4.4.0-1054-snapdragon
gyp ERR! command "/usr/bin/nodejs" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/gruutak/TARCplusplus-iot/node_modules/mraa
gyp ERR! node -v v6.11.0
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok`

arfoll, 2017-06-30:
So that would indicate it's failed the arch check. Which is odd. As a quick
fix install via cmake and it should work.

Gruutak, 2017-06-30:
Done. It installed successfully But how can I proceed to use it in node since it isn't able to find the module?

Propanu, 2017-06-30:
Set `NODE_PATH` to the install location before running node (e.g. `NODE_PATH=/usr/local/lib/node_modules`).

Gruutak, 2017-07-01:
It worked! Thank you very much!