eclipse-mraa/mraa #778: .Net Core / C# bindings

VladimirAkopyan, 2017-07-01:
I am looking for a cross-platform GPIO library that I can use as a basis for a C# applications. This seems to be as close as it gets (or am i wrong?).  Now that .Net is open source we will soon have a release of C# run-time for Arm Linux. 
I want to see if there is any appetite to create bindings for C#, and if anyone would be willing to help.

alext-mkrs, 2017-07-01:
mraa is not cross-platform in a typical sense - it only supports Linux (various distros and kernel versions though), as it uses sysfs extensively, so if you have Windows for IOT in mind, that won't work, at least not without major development effort. There's mock platform, that compiles under MSYS in Windows, but that's a self-contained simulator, not an actual use of Windows system interfaces.

C# bindings for Linux should be doable though and would be quite cool to have, maybe even with the same SWIG used for other bindings.

arfoll, 2017-07-01:
As @alext-mkrs mentions, sounds like you're using cross platform to mean cross board/arch rather than cross OS? If that's correct then mraa is a good choice (unbiased opinion obv ;-)).

How do you feel about swig generated C# bindings? If that would satisfy your need i'm happy to help. If it's manual bindings I'm also happy to include that but probably won't be able to provide all that much help since I know very little about C#...

VladimirAkopyan, 2017-07-03:
yeah, I meant "cross-board". I haven't tried SWIG yet 🚀
I also have not previously used MRAA, is there anything unusual I should be aware of? How's the performance?

mgiaco, 2017-07-05:
Maybe this is interesting also especially for the pur.

http://www.ofitselfso.com/BBBCSIO/BBBCSIO.php

arfoll, 2017-07-05:
Hard to benchmark objectively. Generally I would say perf is pretty good.
Our nodejs bindings are not well liked because they are not asynchronous
but if you look at the python ones performance is really not far off C and
largest overhead is the binding size itself. Most of this is because swig
generated code has alot of redundant checks.

Try it out! It's really the only way. I think generating swig c# bindings
for at least the basic stuff should be relatively easy. We can help you
with the small niggles like getting isrs to work.

alext-mkrs, 2017-07-06:
...and I happen to be working on a different project  which is a cross-platform C# app, so while I'm far off from being an expert, I know a thing or two and would be glad to help as well, having firsthand mraa knowledge.

IvanFarkas, 2018-03-21:
Awesome. Thx @alext-mkrs.

IvanFarkas, 2018-03-21:
@VladimirAkopyan Have you made any progress. I'd like to do the same with swig.

deinok, 2018-09-21:
https://www.nuget.org/packages/SystemExt.Devices.Mraa/

Actualy it requires have libmraa installed, but i'm working on have an aditional NuGet called for example (SystemExt.Devices.Mraa.runtime.linux-arm) that will not require to have libmraa it will use a local compiled version of libmraa

deinok, 2018-09-21:
I'm open to transfer the ownership of the bindings to the MRAA project if they will promise still giving support to .NET

deinok, 2018-09-26:
The lib by itself just make calls to `libmraa.so` using `[DllImport]`.
So in theory, if in a WSL environment you can use libmraa, indirectly you can execute a .NET Core app inside the WSL using this bindings and should work pretty well.
I think that the biggest problem for you will be setup libmraa inside WSL. Once done, it should be pretty easy access to the info using my package