eclipse-mraa/mraa #781: custom BBB yocto image spidev1.1

mgiaco, 2017-07-02:
Hi,

How should my json look like if i want to use my spi device under libmraa. I tried to use spidev and i figured out that it works. So I think it should also work under mraa. My problem is i only have a spidev1.1  
Now I am not sure how the platform entry should look like. So I need bus=1 device=1

```json
  "SPI": [
    { "index":0, "bus": 1, "default": true}
  ]
```
Should I add the miso, mosi... stuff? Because this is already set in the dev tree.

many thanks

mgiaco, 2017-07-03:
@alext-mkrs  hi, many thank. Sure I used the documentation and the examples as a starting point. So from the example here. This means mraa uses spidev0.0 right? and as an example the clock pin is gpio 11 which is definded as lable   {"label": "SPI_CLK"},  so but there is no mapping to a rawpin so the pin is used from the dev tree - right? 

  "SPI": [
    { "index":0, "bus": 0, "slaveselect": 0, "clock": 11, "miso": 7, "mosi": 9, "chipselect": 5, "default": true}
  ],

alext-mkrs, 2017-07-04:
So it goes like this:
1. `bus` is an SPI bus number, i.e. if you have `/dev/spidev5.1`, that would be 5.
1. `slaveselect` is the slave/chip select number from spidev, i.e. in the above case that would be 1. mraa constructs the spidev filename to talk to using "spidev<bus>.<slaveselect>" string.
1. `clock`, `miso`, `mosi` are pins for respective functions, numbers are in pin indexes, not raw pin numbers. These aren't used directly as mraa uses spidev to do the necessary communication, this is more for the accounting.
1. `chipselect` is a pin, which will be used as a "software" CS in GPIO mode, if specified. If provided, should be again in terms of pin indexes, not raw.

mgiaco, 2017-07-04:
@alext-mkrs 
many thanks. I think that should go to the docs.

closed 2017-07-04 by mgiaco

alext-mkrs, 2017-07-05:
Glad to help. Yeah, I've noticed the docs could do with some elaboration and submitted #784 for that