eclipse-mraa/mraa #786: Error: buzzer_play_sound: mraa_pwm_period_us() failed, while executing UPM code example for Grove buzzer sensor with Firmata

priyanka-bagade, 2017-07-07:
While testing Grove Buzzer sensor connected to NUC with Arduino 101, following error occurred:
"buzzer_play_sound: mraa_pwm_period_us() failed"
Hardware setup: Grove Buzzer sensor connected to NUC (Windriver OS) with Arduino 101.
mraa version: 1.7.0
upm version: 1.3.0
Tested code: https://github.com/intel-iot-devkit/upm/blob/master/examples/javascript/buzzer.js
With following lines added for firmata:
"use strict"; 
const mraa = require('mraa');
console.log('MRAA Version: ' + mraa.getVersion()); 
// open connection to firmata
mraa.addSubplatform(mraa.GENERIC_FIRMATA, "/dev/ttyACM0");

Error log:
root@WR-IDP-4993:~# node buzzer.js
MRAA Version: v1.7.0
 
buzzer_play_sound: mraa_pwm_period_us() failed
/root/buzzer.js:54
        console.log( myBuzzer.playSound(chords[chordIndex], 500000) );
                              ^
 
Error: UPM Runtime Error: playSound: buzzer_play_sound() failed
    at Error (native)
    at melody [as _repeat] (/root/buzzer.js:54:31)
    at wrapper [as _onTimeout] (timers.js:275:11)
    at Timer.listOnTimeout (timers.js:92:15)

malikabhi05, 2017-07-07:
The mraa_pwm_write_period() function (called by mraa_pwm_period_us()) does not have a replace function in case of firmata. This leads to the error you see here. This is most likely because the pwm period for firmata is fixed. Working on a fix for this.

mstfldmr, 2017-07-19:
I am having the same error. Any progress?

deadprogram, 2017-09-12:
Any progress on this issue? Thank you very much!

Propanu, 2017-09-12:
Hi @deadprogram this should be fixed now with the latest packages for MRAA/UPM. @priyanka-bagade and @malikabhi05 can confirm.

alext-mkrs, 2018-01-16:
@malikabhi05, I guess it should be long fixed by now, judging by @Propanu's comment. Is it?

Propanu, 2018-01-16:
I remember chatting with @priyanka-bagade about this and it was fixed in the latest MRAA/UPM versions. Closing.

closed 2018-01-16 by Propanu