eclipse-mraa/mraa #787: No ACK signal after readReg(unit8_t reg)

Xenosb, 2017-07-10:
After a read from the slave, master doesn't set ACK bit low. This causes I2C slave to hang.

Master: RPi 3 raspbian + mraa 1.7.0 git build + python2.7
Slave: ATSAMD21 + mbed

To test that communication actually works, I tried writing to a register and this yielded success. After that I tried reading from another register which seemingly succeeds but following reads fail. Here is a simple program and its output.

Master program:
```python
from mraa import I2c

node = I2c(0)
node.frequency(0)
node.address(0x10)

print("write: {}".format(node.writeReg(0xf0, 0x99))) # write: 0
print("read 1: {}".format(node.readReg(0xfa))) # read 1: 16 // correct

# if next row is uncommented, previous row generates "read 1: 255" and next generates Unknown error in mraa.py, line 862.
# print("read 2: {}".format(node.readReg(0xfa)))
```

Since I had a logic analyzer at my hand, I had a look at low level and had something to see. After successful write, and read, master doesn't set the ACK back up. 

[image]

[image]

I also tested on Intel Edison running debian so this shouldn't be platform issue (x86/armhf).

Is this a desired behavior or a bug/issue? I haven't been able to solve this issue on slave side.

Xenosb, 2017-07-10:
Please invalidate this issue.

Rubber-duck hw debugging at it's finest... After this issue report I remembered that logic analyzer has no pull-up resistors and that it might be causing the issue. And yeah, disconnecting it made the program work as I expected.

arfoll, 2017-07-10:
Thanks for coming back with the fix ;-) Good luck with the rest of your project!

closed 2017-07-10 by arfoll