eclipse-mraa/mraa #791: SPI on BBB for Kernel 4.9.x return always zero.

mgiaco, 2017-07-16:
I have some problem with the spi. I am sure i get data because I use a logic analyzer but the receive buffer is always zero.

Is here anyone who using such a new kernel with mraa for a BBB? 

cheers
mathias

simplezhao, 2017-07-20:
Hi, have you solved it?
I have the same question on my linkit 7688, when I write value to the register, but then I always read zero from it.
I use the logic analyzer, and I find it always adds zero after my data.

mgiaco, 2017-07-20:
No sorry I haven´t solved it but I found out it is not only a mraa problem. Because I also have tested with plain spidev from python.  Tomorrow I will also test it with plain C and spidev. Maybe I have some erro in the yocto I use. 

What do you mean with adds zero after my data?

simplezhao, 2017-07-22:
Hi, I use mraa spi to drive rc522 om my linkit 7688, I found bellow:
[image]
the first step , I read the register 0x37, 0x36, 0x02, and it gives me the right answer.
then I write commad to the device, and want to write the value 0x04 to the register 0x02,   From the logic analyzer, i think it works, but i find after I write the value, it writes a zero to the device! I don't do nothing!
finally. I read 0 from the register which I write.

mgiaco, 2017-07-24:
He so I can say spi works on that kernel. My problem was in the device tree. For the BBB core the input line for the sclk must be defined as input. 
@simplezhao  my question was only for the beagle bone black - sorry.

closed 2017-07-24 by mgiaco

arfoll, 2017-07-24:
Thanks for getting back to us!