eclipse-mraa/mraa #796: Error while installing MRAA for node.js

ritesh2014, 2017-08-05:
I am working on Johnny five library that uses Node.js to run scripts. My local machine runs on Windows 10 which is connected to Edison board. To compile the scripts, I need MRAA on my local machine.
When I try to install MRAA on my local machine, I get the below error. Can someone please help how to resolve this.

C:\Program Files\nodejs\node_modules\npm>npm install mraa

> mraa@1.7.0 install C:\Program Files\nodejs\node_modules\npm\node_modules\mraa
> node-gyp rebuild

C:\Program Files\nodejs\node_modules\npm\node_modules\mraa>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "" rebuild )
gyp ERR! configure error
gyp ERR! stack Error: Can't find Python executable "python", you can set the PYTHON env variable.
gyp ERR! stack     at failNoPython (C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\configure.js:449:14)
gyp ERR! stack     at C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\configure.js:404:11
gyp ERR! stack     at C:\Program Files\nodejs\node_modules\npm\node_modules\graceful-fs\polyfills.js:264:29
gyp ERR! stack     at FSReqWrap.oncomplete (fs.js:123:15)
gyp ERR! System Windows_NT 10.0.15063
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\Program Files\nodejs\node_modules\npm\node_modules\mraa
gyp ERR! node -v v6.11.1
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok
npm ERR! Windows_NT 10.0.15063
npm ERR! argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "install" "mraa"
npm ERR! node v6.11.1
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE

npm ERR! mraa@1.7.0 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the mraa@1.7.0 install script 'node-gyp rebuild'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the mraa package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs mraa
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls mraa
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     C:\Program Files\nodejs\node_modules\npm\npm-debug.log

C:\Program Files\nodejs\node_modules\npm>

arfoll, 2017-08-07:
Mraa can't be compiled on windows - however you can use mraaStub for this see - https://github.com/intel-iot-devkit/mraa/tree/master/jsstub

ritesh2014, 2017-08-08:
Thanks. I am using Intel XDK now and its working.

closed 2017-08-08 by ritesh2014