eclipse-mraa/mraa #799: cmake uses the wrong include path for json library

skoehler, 2017-08-11:
On Arch Linux, Raspberry 2, cmake detects the json-c library's include folder as `JSON-C_INCLUDE_DIR=/usr/include/json/` which leads to a rather confusing error

```
[  1%] Building C object src/CMakeFiles/mraa.dir/mraa.c.o
In file included from /usr/include/json/forwards.h:10:0,
                 from /usr/include/json/features.h:10,
                 from /usr/include/bits/libc-header-start.h:33,
                 from /usr/include/stdlib.h:25,
                 from /tmp/mraa-1.7.0/src/mraa.c:32:
/usr/include/json/config.h:9:10: fatal error: string: No such file or directory
 #include <string> //typedef String
          ^~~~~~~~
compilation terminated.
```

I had to override things with `cmake -DJSON-C_INCLUDE_DIR=/usr/include/json-c/`.

The problem is that `/usr/include/json` and `/usr/include/json-c` both contain a file called `json.h` which is used by your cmake scripts to find the right folder. The former belongs to a json c++ library. The latter is the json C library.

The cmake check could probably be improved by using `json_c_version.h` instead of `json.h` for detecting json-c.

alext-mkrs, 2017-08-14:
Thanks for reporting!

alext-mkrs, 2018-06-10:
Looking into this one, I think it can also be fixed by just removing `json` entry @Hbrinj added a while ago at this line: https://github.com/intel-iot-devkit/mraa/blob/master/cmake/modules/FindJSON-C.cmake#L6

@Hbrinj, could you please clarify why you've added that one?