eclipse-mraa/mraa #800: Pwm issue in Nodejs

alaa-eddine, 2017-08-13:
Hello,

When I try to instantiate Pwm in nodejs I get the following error : 

```
var p = new mraa.Pwm(12);
        ^

Error: Illegal arguments for construction of _exports_Pwm
```

I first suspected the Pin id but I tried with Gpio and it's working just fine.

The below code works fine 
```
var mraa = require('mraa'); 
var sleep = require('system-sleep');
var p = new mraa.Gpio(12);
p.dir(mraa.DIR_OUT);
p.write(1);
sleep(1000);
p.write(0);
```

This one crash on constructor line 

```
var mraa = require('mraa'); 
var sleep = require('system-sleep');

var p = new mraa.Pwm(12);
p.enable(true);
sleep(1000);
p.enable(false);
```

Do you have any idea why am I getting this error ?

alext-mkrs, 2017-08-14:
Hi @alaa-eddine, which board is this on?

alaa-eddine, 2017-08-14:
Hi @alext-mkrs 
it's a RPi Zero .... and I just read this mension in the documentation :/

> PWM Whilst the Raspberry Pi is meant to have 1 PWM channel this is currently not supported.

So I suppose it's normal 

Sorry I didn't see that when opening the issue.

is PWM support for raspberry planed ?

alext-mkrs, 2017-08-17:
I see  yes, this is just unsupported. I haven't heard of any plans to add it, patches from you or anyone else are more than welcome!

g-vidal, 2017-11-23:
Hi,
rpi.GPIO lib  enables to add "software PWM"  their capabilities are much lower than the hardware  but they maight help you. It should be a great enhancement if this piece of code from rpi.GPIO could be included in MRAA  :-)

arfoll, 2017-11-23:
Is software PWM really something you want to do? It's incredibly wasteful on power & ressources and essentially unreliable as at higher loads it's just going to get worse and worse.

g-vidal, 2017-11-23:
Hi @arfoll,
I understand and agree with your remarks BUT having just one PWM on a Raspberry Pi is really not sufficient. I have not been able to test the edison 4 PWM (that was ne of the numerous advantages of the Edison and it is really unfortunate that it has been stopped) but it has been discontinued.
 * example 1 some weeks ago e celebrated  the anniversary of colour TV, to be part of the eventI decided to set up an RGB LED to make children uderstand how new TV buid colours. To obtain all colours you definitely need 3 PWM onre for red green and blue. (my jupyter program is on my github commented in french...)
 * example 2 You want to build your own little robot you need   at least 2 PWM for 2 electric motors (2 wheels or 1 wheel and one driving wheel.
Of course for few tens of $ you can get an I2C / PWM  interface  but with you naked Edison or Raspi you are prevented from interesting devs.
Users must be informed of the soft PWM capabilities but if you can it would be great to provide them in MRAA.

arfoll, 2017-11-23:
I understand the need - I'm actually not against merging a patch that does this, I'm just not going to do it myself ;-) I'd happily merge a patch that reuses the same pwm API in mraa, is removable with a conf flag and enables any GPIO port to have soft pwm with the help of decent threading.

<salespitch>A minnowboard max/turbot has 2 HW PWM outputs</salespitch>