eclipse-mraa/mraa #802: GPIO as non-root user on Raspberry Pi doesn't work reliably

skoehler, 2017-08-15:
GPIO based on sysfs does not work reliably for non-root users because of a race between udev and the MRAA application.

Rasbian sets the permission of GPIO-related sysfs files using udev rules. See `/etc/udev/rules.d/99-com.rules` for reference. However, the permissions are set too late. You find a copy of `99-com.rules` here: https://raspberrypi.stackexchange.com/questions/62718/what-to-do-with-99-com-rules. 

The following happens:
mraa writes "5" to /sys/class/gpio/export
mraa writes "out" to /sys/class/gpio/gpio5/direction

However, the second command fails with a Permission Denied error. Retrying the command until the udev rules have taken effect would work. However, I strongly believe this should work the first time as it is currently documented nowhere that the command should be retried.

Also the mraa-gpio utility trips over the same problem. The first time I query a pin, it returns -1, most likely because of the same Permission Denied problem. The second time I query the same pin, it returns a value (0 or 1). This seems to be due to the fact that the gpio pin is never unexported (mraa-gpio does not seem to write "5" to /sys/class/gpio/unexport).

I see the following potential solutions:
- After exporting a gpio pin, MRAA gracefully waits until the permissions of the sysfs files for the GPIO have been adjusted. If some timeout is reached, an error is returned. This should work in most cases, if the Raspberry Pi is not too busy with other things.
- Exporting a GPIO ping in sysfs creates a new device at kernel level. MRAA could be enhanced so that it waits until all udev rules for that new device have been processed. I do not know whether this is possible.
- MRAA switches from using sysfs to using /dev/gpiomem or /dev/gpiochip0 which are the GPIO related devices. I do not know how to do GPIO via these devices or whether it is actually possible. Anyhow, once udev sets the permissions of these devices (e.g. allow write access to gpio group), they stay constant.

skoehler, 2017-08-16:
It seems there is indeed an ioctl based API for doing GPIO via /dev/gpiochip0, see
https://www.kernel.org/doc/Documentation/ABI/testing/gpio-cdev

skoehler, 2017-08-17:
Other libraries (for example the python library [RPi.GPIO](https://pypi.python.org/pypi/RPi.GPIO) or the C library [bcm2835](http://www.airspayce.com/mikem/bcm2835/)) use /dev/gpiomem. However, that seems to be a solution that is specific to the Raspberry Pi and is not very portable. The /dev/gpiochip* interface seems to be part of a future Linux ABI and support for using /dev/gpiochip* in MRAA seems worth while and it solves the permission issues that sysfs implies (There is no standard way of setting sysfs permissions, and Raspbian doing chmod/chown on sysfs files via udev rules seems like an ugly hack).

Propanu, 2017-08-18:
Hi @skoehler, we're actually in the process of adding support for GPIO device nodes on boards that run a new enough kernel. Hard to give an ETA on it at this point though.

skoehler, 2017-08-22:
Is the code publicly available? Can i help testing it, maybe?

arfoll, 2017-08-22:
what we will likely merge is here: https://github.com/stefan-andritoiu/mraa-gpio-chardev

mgiaco, 2017-12-07:
@arfoll any news here?

g-vidal, 2017-12-10:
Just a comment from a non-raspbian user of RaspberryPi with mraa.
According to this [contrib to BB](https://github.com/olegantonyan/adafruit-beaglebone-io-python/commit/6cba967bf2c84444bc5cb7293a62fa3e9393b972); some of the problems may come just from the delay to make the `/sys/class/gpio/gpio#` available for the rest of the program. Would it be possible to make a light amendment introducing a pause/check in that way ? The problems I am facing  seem to fall into that category. Here is  my rules file (not clean but operational) : 
```shell
SUBSYSTEM=="gpio*", PROGRAM="/bin/sh -c 'chown -R root:gpio /sys/class/gpio && chmod -R 770 /sys/class/gpio; \
chown -R root:gpio /sys/devices/virtual/gpio && chmod -R 770 /sys/devices/virtual/gpio; \
chown -R root:gpio /sys/devices/platform/soc/3f200000.gpio && chmod -R 770 /sys/devices/platform/soc/3f200000.gpio; \
chown -R root:gpio /sys/devices/platform/soc/soc:virtgpio && chmod -R 770 /sys/devices/platform/soc/soc:virtgpio; \
chown -R root:gpio /sys/bus/platform/drivers/pinctrl-bcm2835 && chmod -R 770 /sys/bus/platform/drivers/pinctrl-bcm2835 '"

SUBSYSTEM=="bcm2835-gpiomem", KERNEL=="gpiomem", GROUP="gpio", MODE="660"
SUBSYSTEM=="gpio", KERNEL=="gpiochip*", ACTION=="add", PROGRAM="/bin/sh -c 'chown root:gpio /sys/class/gpio/export /sys/class/gpio/unexport ; \
chmod 660 /sys/class/gpio/export /sys/class/gpio/unexport'"
SUBSYSTEM=="gpio", KERNEL=="gpio*", ACTION=="add", PROGRAM="/bin/sh -c 'chown root:gpio /sys%p/active_low /sys%p/direction /sys%p/edge /sys%p/value ; \
chmod 660 /sys%p/active_low /sys%p/direction /sys%p/edge /sys%p/value'"
```
Here are the permissions obtained on the gpio at boot : 
 ```shell
 ll /sys/class/gpio/
total 0
drwxrwx---  2 root gpio    0 oct.   6 10:18 ./
drwxr-xr-x 49 root root    0 janv.  1  1970 ../
-rw-rw----  1 root gpio 4,0K oct.   6 10:18 export
lrwxrwxrwx  1 root gpio    0 oct.   6 10:18 gpiochip0 -> ../../devices/platform/soc/3f200000.gpio/gpio/gpiochip0/
lrwxrwxrwx  1 root gpio    0 oct.   6 10:18 gpiochip100 -> ../../devices/gpiochip2/gpio/gpiochip100/
lrwxrwxrwx  1 root gpio    0 oct.   6 10:18 gpiochip128 -> ../../devices/gpiochip1/gpio/gpiochip128/
-rw-rw----  1 root gpio 4,0K oct.   6 10:18 unexport
```
Here is a small python program : 
```shell 
cat led33.py 
import mraa, time
led = mraa.Gpio(33)
led.dir(mraa.DIR_OUT)
led.write(1)
time.sleep(2)
led.write(0)
```
The execution fails with a permission denied message (note thet the the GPIO number translation works perfectly) :
```shell
Dec 10 22:20:20 3littlehills libmraa[1311]: libmraa version v1.8.0-17-g02963b7 initialised by user 'vidal' with EUID 1005
Dec 10 22:20:20 3littlehills libmraa[1311]: libmraa initialised for platform 'Raspberry Pi 3 Model B' of type 5
Dec 10 22:20:20 3littlehills libmraa[1311]: gpio13: dir: Failed to open 'direction' for writing: Permission denied
Dec 10 22:20:20 3littlehills libmraa[1311]: gpio13: Failed to open 'value': Permission denied
Dec 10 22:20:22 3littlehills libmraa[1311]: gpio13: write: Failed to write to 'value': Operation not permitted
```
While the direct python console execution works perfectly with **the same user**  : 
```shell
 python
Python 2.7.14 (default, Sep 17 2017, 18:50:44) 
[GCC 7.2.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import mraa, time
>>> led = mraa.Gpio(33)
>>> led.dir(mraa.DIR_OUT)
0
>>> led.write(1)
0
>>> led.write(0)
0
```
Surprisingly if we execute again led33.py **after**  the shell execution It works!! just because the `sys/class/gpio` file exists : 
```shell
  ll /sys/class/gpio/
total 0
drwxrwx---  2 root gpio    0 déc.  10 22:24 ./
drwxr-xr-x 49 root root    0 déc.  10 22:10 ../
-rwxrwx---  1 root gpio 4,0K oct.   6 10:18 export*
lrwxrwxrwx  1 root gpio    0 déc.  10 22:24 gpio13 -> ../../devices/platform/soc/3f200000.gpio/gpiochip0/gpio/gpio13/
lrwxrwxrwx  1 root gpio    0 oct.   6 10:18 gpiochip0 -> ../../devices/platform/soc/3f200000.gpio/gpio/gpiochip0/
lrwxrwxrwx  1 root gpio    0 oct.   6 10:18 gpiochip100 -> ../../devices/gpiochip2/gpio/gpiochip100/
lrwxrwxrwx  1 root gpio    0 oct.   6 10:18 gpiochip128 -> ../../devices/gpiochip1/gpio/gpiochip128/
-rwxrwx---  1 root gpio 4,0K oct.   6 10:18 unexport*
```
Of course using sudo works but the aim is to use GPIOs without root... Any help or suggestion where to put the counter to check if it works?
Thanks for your work.

skoehler, 2017-12-10:
> Just a comment from a non-raspbian user of RaspberryPi with mraa.
According to this contrib to BB; some of the problems may come just from the delay to make the /sys/class/gpio/gpio# available for the rest of the program. Would it be possible to make a light amendment introducing a pause/check in that way ? The problems I am facing seem to fall into that category. Here is my rules file (not clean but operational)

Abusing udev to adjust the permissions of files in /sys is a bad idea. There is no guarantee when the permissions of these files are set. I'm not even sure whether mraa could somehow communicate with udev to wait until the permissions have been set. Settings the permissions of the gpio files in /sys seems like a (bad) workaround to me, that was implemented simply because there was no general way to access Gpio. 

A clean solution is around the corner, where mraa will access the Gpio pins via /dev/gpiochip*. Controlling the permissions of device in /dev via udev is a common practice and will work without issues.

The easiest workaround is to simply retry setting the direction of a gpio pin until it succeeds. I'm pretty sure the mraa devs wont waste time on resolving the access via sysfs if access via /dev is around the corner.

arfoll, 2017-12-11:
@skoehler well actually we kinda have both. Imraa will set permissions in sysfs (which as you say is nasty but - it works :)). We also have a gpiochip access still not quite in master check the branch here: https://github.com/intel-iot-devkit/mraa/tree/chardevmerge2

g-vidal, 2017-12-12:
@arfoll I checked the code from the branch :-)
compilation is OK but either noting happens if used with sudo or various errors if used without sudo on [GPIOD_INTERFACE].
I suppose things are not settled, I'll wait for the merge but if you need some testing and return info let me know.
Thanks again