eclipse-mraa/mraa #805: Sending more than 32 bytes with I2c write

ukBaz, 2017-08-22:
Hello,

I am using the Python bindings of mraa to access an OLED display on the I2C bus.

To update the screen as fast as possible I want to send a 'screen load' of data in one I2c.write command. This is 1024 bytes. However only 32 bytes can be sent at one time which appears to be something coded into the library with I2C_SMBUS_I2C_BLOCK_MAX.
I understand that 32 bytes is an SMBUS limitation not I2C.

Is 32 bytes limit something that a user can change?

Thanks

arfoll, 2017-08-22:
So generally no - there's no easy way for a user to change this - we use the I2C_SMBUS APIs so we follow the SMBUS limitations. This isn't really intended however. We use the SMBUS commands which really are a subset, and so end up with these limitations. It would be nice to use pure i2c, I'm not actually sure there's a reason why we don't, it looks from looking here https://www.kernel.org/doc/Documentation/i2c/dev-interface that there should be no issues with boards supporting SMBUS but not I2c but it remains to be seen.

I'd happily take a patch for this and make sure it's tested on a number of boards, but don't have time to do an implementation any time soon.

ukBaz, 2017-08-27:
Sadly I am not in a position to do a patch either.

mgiaco, 2017-09-04:
https://stackoverflow.com/questions/25982525/why-i2c-smbus-block-max-is-limited-to-32-bytes

So maybe this could be implemented in mraa write function.

Mani-Sadhasivam, 2017-11-02:
@ukBaz In Dragonboard410c there is a quirk in I2C controller IP. It can support only the max of 256 bytes per transfer. So, even if you use raw I2C interface there is no way to transfer 1024 bytes in one shot.

Mani-Sadhasivam, 2018-02-16:
Hi @ukBaz,

I have added a patch for fixing the SMBUS limitation in `mraa_i2c_write` call. Can you please test it?

Commit: https://github.com/Mani-Sadhasivam/mraa/commit/24f910fe2b4439fc3674c3bd2bd512485bf7fc03

You can just clone and build my fork.Essentially, I just replaced the SMBUS interface with raw I2C for this function. However, as I said before, Dragonboard410c I2C controller has its own limitation of 256 bytes.

ukBaz, 2018-02-17:
Thanks for looking at this. I hope to get to this soon

arfoll, 2018-02-19:
@Mani-Sadhasivam we could merge something like this, let's just make a mraa_boolean_t is_smbus property and use the i2c paths by default. That way we can just test boards that are real i2c and slowly remove the boards using the smbus paths.

Mani-Sadhasivam, 2018-02-19:
@arfoll Using SMBUS has its own advantage and most of the IP's support it. IMO we _should_ not use raw I2C for all of our `mraa_i2c` transfers. But we can use it in places where SMBUS imposes limitations, like in `mraa_i2c_write`

arfoll, 2018-02-19:
Maybe actually maybe just a size check would make more sense and just upgrade to i2c if it's too big for an smbus transaction? Tbh I have to re-read what the exact differences are.

Mani-Sadhasivam, 2018-02-19:
@arfoll 

> Maybe actually maybe just a size check would make more sense and just upgrade to i2c if it's too big for an smbus transaction

Sounds good!

ukBaz, 2018-02-25:
Hi @Mani-Sadhasivam ,

I have had a look at this today without any joy. If I run with the 32byte chunk size I am able to send data to the screen. 

This is my function that is sending the data: 
```python
    def _send_data32(self, data):
        chunk_size = 32
        for i in range(0, len(data), chunk_size):
            frame = [OLED96.SEND_DATA] + data[i:i + chunk_size]
            self.device.write(bytearray(frame))
```
If I set the chunk size to 32 or smaller then I can see screen update. The smaller the chunk size the slower the update.

If I go larger than 32 then the hardware seems to freeze and I have reboot the hardware. 
There doesn't seem to be any obvious message or error thrown.

Can I just check that I am using the correct procedure to get and build the fix?
```linux
mkdir mraa
cd mraa
git clone https://github.com/Mani-Sadhasivam/mraa.git
cd mraa
git checkout i2c_raw
git checkout 24f910fe2b4439fc3674c3bd2bd512485bf7fc03
mkdir build
cd build/
cmake -DBUILDSWIG=ON \
      -DBUILDSWIGPYTHON=ON \
      -DBUILDSWIGNODE=OFF \
      -DBUILDSWIGJAVA=OFF \
      -DCMAKE_INSTALL_PREFIX=/usr \
      ..
make
sudo make install
```
A working syslog looks like:
```
Feb 25 17:26:30 linaro-alip libmraa[2265]: libmraa version v1.6.1-162-g24f910f initialised by user 'root' with EUID 0
Feb 25 17:26:30 linaro-alip libmraa[2265]: libmraa initialised for platform 'DB410C' of type 9
Feb 25 17:26:30 linaro-alip libmraa[2265]: i2c_init: Selected bus 0
```

A failing syslog looks like:
```
Feb 25 17:27:44 linaro-alip libmraa[2282]: libmraa version v1.6.1-162-g24f910f initialised by user 'root' with EUID 0
Feb 25 17:27:44 linaro-alip libmraa[2282]: libmraa initialised for platform 'DB410C' of type 9
Feb 25 17:27:44 linaro-alip libmraa[2282]: i2c_init: Selected bus 0
```

Mani-Sadhasivam, 2018-02-26:
Hi @ukBaz ,

Thanks for testing out the patch and the procedure is correct. However, the behaviour you mentioned looks weird. Will investigate more and get back.