eclipse-mraa/mraa #810: Python GPIO Pullup on Dragonboard

ukBaz, 2017-08-28:
I have tried to do a pull-up on a GPIO on the Dragonboard 410c and it does not seem to have worked.

With nothing attached to the GPIO I have done the following experiment:

```
root@linaro-alip:~# python3
Python 3.5.3 (default, Jan 19 2017, 14:11:04) 
[GCC 6.3.0 20170118] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import mraa
>>> mraa.getVersion()
'v1.7.0'
>>> mraa.getPlatformName()
'DB410C'
>>> mraa.getPinName(30)
'GPIO-H'
>>> btn = mraa.Gpio(30)
>>> btn.dir(mraa.DIR_IN)
0
>>> btn.read()
0
>>> btn.mode(mraa.MODE_PULLUP)
7
>>> btn.read()
0
```

Any suggestions as to what I'm doing wrong?

arfoll, 2017-08-29:
@Mani-Sadhasivam maybe have more details on this and why it doesn't work.

Mani-Sadhasivam, 2017-08-30:
@ukBaz Sorry, pull up mode is not exposed through sysfs in Dragonboard410c (which means no direct control over it). AFAIK pull up/down modes are only exposed by **some** Intel SBC's like Galileo, hence mraa supports it. 

After executing the above mentioned python code, you can see the log in '/var/log/syslog', it would've logged the output error opening 'drive' under sysfs.

It was my mistake commenting on the forum about this support. I'll take back my statement.

Anyway, thanks for pointing this out.

alext-mkrs, 2017-09-08:
While not strictly a "hardware issue" I'm putting the label to denote this is board-specific and outside of mraa's control. As looks like there's nothing more to be done here, I'm closing this one.

closed 2017-09-08 by alext-mkrs