eclipse-mraa/mraa #815: [up board] All of GPIO Pins can not be used

wanghongjuan, 2017-09-06:
Hello,

I am use this [up board](http://www.up-board.org/up/) with Ubuntu 16.04,  but blocked when invoke GPIO with mraa (javascript version),  [test code](https://up-community.org/wiki/MRAA/UPM#Example_1:_Blink_an_LED_on_a_GPIO_output_pin) as below:
```
led.js

var m = require('mraa'); //require mraa

var myLed = new m.Gpio(7); //LED hooked up to digital pin 7
myLed.dir(m.DIR_OUT);      //set the gpio direction to output
var ledState = true;       //Boolean to hold the state of Led

function periodicActivity()
{
    myLed.write(ledState?1:0);           //if ledState is true then write a '1' (high) otherwise write a '0' (low)
    ledState = !ledState;                //invert the ledState
    setTimeout(periodicActivity,500);    //call the indicated function after 1 second (1000 milliseconds)
}

periodicActivity(); //call the periodicActivity function
```
Error occurred when running `node led.js` whether using role admin(root) or general user:
[image]

Journalctl log as below:
[image]

I tried all pins as this [mapping](https://up-community.org/wiki/Pinout), but all of them got same error as above.

Node.js Version: V6.11.2
Npm Version: V3.10.10

arfoll, 2017-09-06:
Looks like a permission issue to me - try it as rot - you're getting permission denied.

wanghongjuan, 2017-09-06:
Thanks for your quickly response, I tried it as root, unfortunately, got same result as above

arfoll, 2017-09-06:
@nicolalunghiemutex any ideas? Do you know if there's a bios issue or something on up1 that would cause this? seems liek gpio4 just won't export itself from sysfs

nicolalunghiemutex, 2017-09-06:
Dear @wanghongjuan

1. what kernel are you using? (uname -a)
2. can you try do execute the following command:
   `sudo -i`
   `echo 4 > /sys/class/gpio/export`
  and report back the result?
2. can you try do execute the following command:
sudo ls /sys/kernel/debug/pinctrl
and report the result?

Best Regards
Nicola Lunghi

wanghongjuan, 2017-09-07:
Hi, @nicolalunghiemutex thanks for you reply, please see below information:
1.  The kernel that I am using is:
```
$ uname -a
Linux up2 4.10.0-33-generic #37~16.04.1-Ubuntu SMP Fri Aug 11 14:07:24 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
```
2. The result is:
```
root@up2:~# echo 4 > /sys/class/gpio/export
-bash: echo: write error: Invalid argument
```
3. The result is:
```
yunfei@up2:~$ sudo ls /sys/kernel/debug/pinctrl
INT33FF:00  INT33FF:01  INT33FF:02  INT33FF:03  pinctrl-devices  pinctrl-handles  pinctrl-maps
```

For the up board kernel, I will install it later.

nicolalunghiemutex, 2017-09-07:
Dear @wanghongjuan
without the Up Linux Kernel the GPIO on the board simply doesn't work.
Please install our kernel from the PPA then try again.

Regards
Nicola Lunghi

alext-mkrs, 2017-09-30:
Closing for no activity and solution provided, feel free to comment if you need to reopen it.

closed 2017-09-30 by alext-mkrs

Jianfengusa, 2021-06-26:
Hi I am having a similar issue, 
[image]

I believe that I am using upboard kernel, can someone help?

[image]