eclipse-mraa/mraa #817: mraa_gpio_lookup return value

NickBridge, 2017-09-08:
Hi,

according to "mraa_gpio_lookup" function declaration in "common.h" file, return statement should be:

return i;

and not:

return plat->pins[i].gpio.pinmap;

Is it right?

alext-mkrs, 2017-09-08:
Hmm, looks indeed like a bug to me. @vintummala, could you please comment, as the author of that piece of code?

vintummala, 2017-09-08:
The intent of mraa_gpio_lookup() is to pass the name of GPIO pin and get the corresponding MRAA index number for it.

In peripheralman.c, we get the gpio names and assign the MRAA index number to pinmap for the names. This is done in mraa_peripheralman_plat_init() api.
Returning the same in mraa_gpio_lookup().

alext-mkrs, 2017-09-09:
I see, thanks. Well, it looks like this is in contradiction with what `pinmap` field is used everywhere else. This field normally contains the OS-level pin number, while something that we would call "mraa pin index" is namely the index within the pin table, i.e. `i` in this case. As an example - `mraa_gpio_init()` uses namely that as an input and `pinmap` is used only in the `_internal()` part of the init, to actually export the GPIO at the OS level.

I see that in `mraa_peripheralman_plat_init()` the value of `i` is being assigned to `pinmap` actually, but that is a sort of a corner case and doesn't really justify using `pinmap` as the "mraa pin index`.

In addition, just above the `mraa_gpio_lookup()` we have a reverse function - `mraa_get_pin_name()`, where again the input parameter is used namely as a pin table index, there's no search within `pinmap`. And that's a correct behavior.

So looks like we either need to create a proper `_lookup()` function, which would use the pin table index + make the current `_lookup()` Peripheral Manager-specific, or change the Peripheral Manager-related piece to actually use the pin index + correct the `_lookup()` to use it as well. @arfoll, any ideas?

alext-mkrs, 2017-09-30:
@arfoll, please? :)

alext-mkrs, 2018-01-16:
@arfoll, @propanu, what do you think of this one, given @vintummala's explanation and my investigation above?

Propanu, 2018-01-17:
Since we're talking about common functionality here, I would prefer to keep the `_lookup()` functions generic and not tied to a specific platform. In fact, even @vintummala agrees to this and I quote:

> The intent of mraa_gpio_lookup() is to pass the name of GPIO pin and get the corresponding MRAA index number for it.

So we're talking MRAA pin numbers not the OS pin numbers used internally. I would recommend option 2, and patching up the Peripheral Manager code accordingly.

arfoll, 2018-01-18:
This makes alot of sense (and sorry for being absent for so long on this) - this is a good change to add to the v2 list

commit 2d210621 (2018-02-07): mraa.c: make mraa_gpio_lookup() return proper pin index

Pinmap field value is an OS pin number, not mraa pin index,
so make the function return the proper one.

Fixes #817.
closed 2018-02-07 by alext-mkrs