eclipse-mraa/mraa #821: RE: cmake ..

silver2row, 2017-09-15:
Hello,

I tried the cmake .. cmd and came up empty. What am I supposed to type with cmake ..?

Seth

P.S. The newer version of our BBGW image file/Debian Stretch Distro has nodejs v6.11.2 has nodejs-dev already installed but I get an error code with it stating that nodejs-dev will not be installed.

silver2row, 2017-09-15:
Hello,

I tried to download the nodejs version of MRAA on my BBGW. I have met issues before. So now, I am using the Ubuntu apt-get to get MRAA.

Seth

P.S. I did a mkdir build and cd build. I then tried to apply cmake mraa .. and an error came up with cmake. I must not have cmake on my BBGW. Who knows anything about compiling MRAA on the BBGW?

silver2row, 2017-09-15:
**bash: cmake: command not found** is what I found on my system when I tried building in directory build.

arfoll, 2017-09-15:
You need to install cmake (apt-get install cmake) and then run ```cmake ..``` from the build/ folder. Have a look at docs/building.md there's a bunch of details/info in there.

silver2row, 2017-09-21:
Hello...thank you. I needed a push in the water. Thank you for the starting advice. I will return here if things go incorrectly. 

Seth

P.S. I am going to try to do the bidding tonight. I will get back here with a yay or nay.

silver2row, 2017-09-21:
Hello Again...incorrect is my middle name. 

Here:

The following packages have unmet dependencies:
nodejs-dev : Depends: nodejs (= 4.8.2~dfsg-1) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

Also:

CMake Error: The source directory "/home/debian" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.

Seth

P.S. If you know what I have come across that is incorrect, please let me know.

silver2row, 2017-09-21:
Oh and...

I can run this cmd: sudo mraa-gpio. Before, I could not run this cmd. 

Now, it shows my list of GPIO and AIO/etc...

Seth

P.S. Should this clue me in on something or not?

arfoll, 2017-09-21:
So I have no idea why you are getting package manager errors around node.js or things about /home/debian - maybe you're just in the wrong place?, but if ```mraa-gpio list``` is showing you good stuff then essentially libmraa.so is working correctly. Maybe the best way to start would be to use the python bindgs? Go to build/src/python/python2/ and run ```python```. Then in the shell start doing things like:
```
import mraa as m
m.getPlatformName()
```

Have a look in examples/python/ for more things!

silver2row, 2017-09-22:
Okay,

I will try out the **import mraa** and so on to see if Python is recognizing it.

Seth

silver2row, 2017-09-22:
Hello arfoII,

Seth here. I tried to import the mraa as m and got further in my software. I also got m.getPlatformName() in and got the "name" of my board.

It stated BeagleBone Black Rev. B instead of what I have but it works. Good on you. Thank you, again.

Seth

closed 2017-09-22 by silver2row