eclipse-mraa/mraa #822: RS485 - UART

ar0n, 2017-09-20:
I have an IOT2020, identified as "'Intel Galileo Gen 2" by libmraa.

I'm trying to use a [MAX487](https://datasheets.maximintegrated.com/en/ds/MAX1487-MAX491.pdf) with this, sending data works receiving is not. The problem is that i use a GPIO pin to control the transmitters direction. Here is my code:

```
	const unsigned char packet[4] = {0xa0, 0x01, 0xff, 0xff};
	do {
		mraa_gpio_write(gpio, 1);
		mraa_uart_write(uart, packet, 4);
		mraa_uart_flush(uart);
		mraa_gpio_write(gpio, 0);
		while(mraa_uart_data_available(uart, 1000)) {
           		mraa_uart_read(uart, &buf, 1);
           		printf("%.2X ", buf);
           		fflush(stdout);
           	}
		printf("\n");
	} while (1);
```
However the gpio is fast enough mraa_uart_flush/tcdrain takes too long.
Sending takes ~4ms after that tcdrain blocks for ~9ms. [Scope image](http://aron.ws/scope.png)
So I get no response or sometimes partial responses over the rs485 bus.

Usually the RTS line of the serial port is used to enable sending, but there is no RTS line defined on this platform in the source, however mraa_uart_set_flowcontrol return with success.

Is there a way to solve this?

jan-kiszka, 2017-09-20:
There are slow (I2C multiplexer based) and faster GPIOs (SoC-based, direct register access) on the platform. I don't know by heart which are which, but the mraa gpio tool should tell you.

arfoll, 2017-09-20:
As Jan says pins 0, 1, 2, 3, 10 & 12 should be faster if you enable
gpio_mmap. at least from a quick look at the sources, the mraa-gpio tool
will tell you.

mraa_uart_set_flowcontrol() will be succesful if it's able to be set from
the kernel but since there's no pin for this, it's not very helpful. It
would work on a normal gen2 using the uart on the debug controller (w/e the
FTDI chip) but not sure if that is an option and/or if the board you have
has this.

ar0n, 2017-09-20:
Yeah I'm using this:

```
	if (mraa_gpio_use_mmaped(gpio, 1) != MRAA_SUCCESS) {
		fprintf(stdout, "mmapped access to gpio %d not supported, falling back to normal mode\n", 2);
	}
```
Digital pin 02 is memory mapped. 
And I've seen performance increase, whe running a simple:

```
while(1) {
   mraa_gpio_write(gpio, 1);
   mraa_gpio_write(gpio, 0);
}
```
gives me a ~550KHz signal, thats fast enough...

So the problem is not the slowness of the gpio. The problem is that mraa_uart_flush takes too long, even if the buffer is empty.

So doing this, without writing anything to the uart:

```
while(1) {
   mraa_gpio_write(gpio, 1);
   mraa_uart_flush(uart);
   mraa_gpio_write(gpio, 0);
}
```
i get a 190 KHz signal. Thats ~3 times slower.

Here is my full source: https://pastebin.com/gcbXtdL1

alext-mkrs, 2017-09-20:
That part is outside of mraa - as you can see in [uart.c](https://github.com/intel-iot-devkit/mraa/blob/c6b1b18a9d9c4c633b85433b1974387225c480c9/src/uart/uart.c#L448-L466) we just call `tcdrain()` for this one and then it's up to the OS. Do you really need to do the flush?

ar0n, 2017-09-21:
I have to make sure, the data has been transmitted. Is there any way to know if the transmit buffer is empty ?

alext-mkrs, 2017-09-23:
Not in mraa, maybe on the OS level, depending on the UART driver you have (and whether it exposes anything in this regard). Basically, mraa does nothing special about UARTs - it just wraps the OS services and API for them.

ar0n, 2017-09-23:
I've added a monostable multivibrator to controll the transceivers direction. The GPIO is not good for RT stuff, anyway. Thx for the help!

closed 2017-09-23 by ar0n