eclipse-mraa/mraa #823: cmake ..

silver2row, 2017-09-21:
Hello,

I am trying to build with cmake and it is not working. I have a BeagleBone Green Wireless. I can access the sudo mraa-gpio list cmd in the terminal but I can only do that one cmd.

I have some issues. I installed cmake and I do not know if the sudo mraa-gpio list cmd is a key to knowing if the build worked. 

Seth

silver2row, 2017-09-21:
Hello Again,

I will have the error messages posted soon. Sorry for the delay.

Seth

silver2row, 2017-09-21:
Hello...my error codes are located here: https://github.com/intel-iot-devkit/mraa/issues/821. 

I accidentally erased the post and then reopened it. Sorry for the oddness.

Seth

P.S. Let a brother know what is going on during my build. If you need any info, just let me know.

arfoll, 2017-09-21:
I'll close this one - let's carry on in #821

closed 2017-09-21 by arfoll