eclipse-mraa/mraa #825: Platform Loading: Undefined symbol mraa_i2c_init

nigeltiany, 2017-09-25:
```json
{
    "version": "0.0.1",
    "platform": [
        {
            "id": 1010,
            "name": "Odroid",
            ...
            "uart_count": 1,
            "i2c_count": 2,
            ...
        }
    ],
    "layout": [
        {"label": "GPIO-4"},
        ...
        {"label": "X-GPIO-12"}
    ],
    "GPIO": [
        { "index": 0, "rawpin": 173 },
       ...
        { "index": 22, "rawpin": 187 },
        { "index": 23, "rawpin": 34 },
        { "index": 24, "rawpin": 188 },
        ...
        { "index": 29, "rawpin": 228 }
    ],
    "I2C": [
        { 
            "index": 0,
            "sdapin": 11,
            "sclpin": 9,
            "default": false
        },
        {
            "index": 1,
            "sdapin": 22,
            "sclpin": 24,
            "default": true
        }
    ],
    "UART": [
       ...
    ]
}
```
```bash
Sep 25 21:17:16 odroid libmraa[2844]: libmraa version v1.8.0 initialised by user 'odroid' with EUID 1000
Sep 25 21:17:16 odroid libmraa[2844]: init_json_platform: AIO wasn't found in the json file
Sep 25 21:17:16 odroid libmraa[2844]: init_json_platform: SPI wasn't found in the json file
Sep 25 21:17:16 odroid libmraa[2844]: init_json_platform: No bus specified for I2C at position: 0
Sep 25 21:17:16 odroid libmraa[2844]: init_json_platform: No bus specified for I2C at position: 1
Sep 25 21:17:16 odroid libmraa[2844]: init_json_platform: No rawpin specified for UART at position: 0
Sep 25 21:17:16 odroid libmraa[2844]: init_json_platform: PWM wasn't found in the json file
Sep 25 21:17:16 odroid libmraa[2844]: init_json_platform: Platform Odroid initialised via json
Sep 25 21:17:16 odroid libmraa[2844]: libmraa initialised for platform 'Odroid' of type 97
Sep 25 21:18:39 odroid sudo[2865]:   odroid : TTY=pts/2 ; PWD=/home/odroid/Documents/drone-v2 ; USER=root ; COMMAND=/usr/bin/node index
Sep 25 21:18:39 odroid sudo[2865]: pam_unix(sudo:session): session opened for user root by odroid(uid=0)
```
```json
{
    "version": "0.0.1",
    "platform": [
        {
            "id": 1010,
            "name": "Odroid",
            "pin_count": 30,
            "gpio_count": 30,
            "aio_count": 0,
            "uart_count": 1,
            "i2c_count": 2,
            "spi_count": 0,
            "pwmDefPeriod": 0,
            "pwmMaxPeriod": 0,
            "pwmMinPeriod": 0
        }
    ],
    "layout": [
        {"label": "GPIO-4"},
        ...
        {"label": "X-GPIO-12"}
    ],
    "GPIO": [
        { "index": 0, "rawpin": 173 },
       ...
        { "index": 22, "rawpin": 187 },
        { "index": 23, "rawpin": 34 },
        { "index": 24, "rawpin": 188 },
        ...
        { "index": 29, "rawpin": 228 }
    ],
    "I2C": [
        { 
            "index": 0,
	    "bus": 1,
            "sdapin": 11,
            "sclpin": 9,
            "default": false
        },
        {
            "index": 1,
	    "bus": 5,
            "sdapin": 22,
            "sclpin": 24,
            "default": true
        }
    ],
    "UART": [
        {
            "index": 0,
            "chipID": 1010,
	    "rawpin": 172,
            "rx": 2,
            "tx": 4,
            "path": "/dev/ttySACI0",
            "default": true
        }
    ]
}
```
```bash
Sep 25 21:18:39 odroid libmraa[2869]: libmraa version v1.8.0 initialised by user 'root' with EUID 0
Sep 25 21:18:39 odroid libmraa[2869]: Unknown Platform, currently not supported by MRAA
Sep 25 21:18:39 odroid libmraa[2869]: libmraa initialised for platform 'Unknown platform' of type 98
Sep 25 21:18:39 odroid libmraa[2869]: init_json_platform: AIO wasn't found in the json file
Sep 25 21:18:39 odroid libmraa[2869]: init_json_platform: SPI wasn't found in the json file
Sep 25 21:18:39 odroid libmraa[2869]: init_json_platform: PWM wasn't found in the json file
Sep 25 21:18:39 odroid libmraa[2869]: init_json_platform: Platform Odroid initialised via json
Sep 25 21:18:39 odroid sudo[2865]: pam_unix(sudo:session): session closed for user root
Sep 25 21:19:08 odroid sudo[2755]: pam_unix(sudo:session): session closed for user root
Sep 25 21:19:15 odroid libmraa[2894]: libmraa version v1.8.0 initialised by user 'odroid' with EUID 1000
Sep 25 21:19:15 odroid libmraa[2894]: init_json_platform: AIO wasn't found in the json file
Sep 25 21:19:15 odroid libmraa[2894]: init_json_platform: SPI wasn't found in the json file
Sep 25 21:19:15 odroid libmraa[2894]: init_json_platform: PWM wasn't found in the json file
Sep 25 21:19:15 odroid libmraa[2894]: init_json_platform: Platform Odroid initialised via json
Sep 25 21:19:15 odroid libmraa[2894]: libmraa initialised for platform 'Odroid' of type 97
...
Sep 25 21:20:23 odroid sudo[2966]:   odroid : TTY=pts/2 ; PWD=/home/odroid/Documents/drone-v2 ; USER=root ; COMMAND=/usr/bin/node index
Sep 25 21:20:23 odroid sudo[2966]: pam_unix(sudo:session): session opened for user root by odroid(uid=0)
Sep 25 21:20:23 odroid libmraa[2970]: libmraa version v1.8.0 initialised by user 'root' with EUID 0
Sep 25 21:20:23 odroid libmraa[2970]: Unknown Platform, currently not supported by MRAA
Sep 25 21:20:23 odroid libmraa[2970]: libmraa initialised for platform 'Unknown platform' of type 98
Sep 25 21:20:23 odroid libmraa[2970]: init_json_platform: AIO wasn't found in the json file
Sep 25 21:20:23 odroid libmraa[2970]: init_json_platform: SPI wasn't found in the json file
Sep 25 21:20:23 odroid libmraa[2970]: init_json_platform: PWM wasn't found in the json file
Sep 25 21:20:23 odroid libmraa[2970]: init_json_platform: Platform Odroid initialised via json
Sep 25 21:20:23 odroid sudo[2966]: pam_unix(sudo:session): session closed for user root
```
```javascript
"use strict";
const mraa = require('mraa'); //require mraa
mraa.initJsonPlatform('./odroid.json');

var pca9685 = require('jsupm_pca9685');
var pca9685Driver = new pca9685.PCA9685(5, 0x45);
```
```bash
odroid@odroid:~/Documents/untitled$ sudo node index
node: symbol lookup error: /home/odroid/Documents/untitled/node_modules/jsupm_pca9685/build/Release/jsupm_pca9685.node: undefined symbol: mraa_i2c_init
```

Propanu, 2017-09-25:
Did you install `jsupm_pca9685` from npm? If so, make sure node-gyp knows where `mraa.node` is when compiling the upm module.

nigeltiany, 2017-09-25:
Yes the module is installed. How do I make sure node-gyp knows where
mraa.node is ?

nigeltiany, 2017-09-28:
Besides, ```mraa >= 1.0.0``` is already listed as a dependency why would it not find the mraa module and its contents including mraa.node

Propanu, 2017-09-28:
The jsupm modules from npm actually don't list mraa as a dependency because that was causing redundant mraa installs when using a global install. We'll eventually change that and switch to a better flow but haven't had the chance to do so yet. After you install mraa, set the `NODE_PATH` variable to the location where you installed it.

nigeltiany, 2017-09-28:
I have copied my built mraa to ```/usr/ib/node_modules/mraa``` and my ```NODE_PATH``` variable is ```NODE_PATH=/usr/lib/nodejs:/usr/lib/node_modules:/usr/share/javascript:/usr/lib/node_modules/mraa:/usr/lib/node_modules/mraa/mraa.node```

Propanu, 2017-09-28:
If you're using your own built mraa, you'll have to build `jsupm_pca9685` module with cmake too. The reason is, npm mraa is statically linked and has all the symbols, while the one built with cmake links to `libmraa.so`. In turn, `jsupm_pca9685` from npm will only link against `mraa.so` and doesn't care about `libmraa.so` which has the init function. Hope this clarifies it. Let us know if you need help building the upm library, it should be very similar to mraa.

nigeltiany, 2017-09-28:
Thanks for all the help. The module seems to work now when built from source. New Issue,
 ```shell
var pca9685Driver = new pca9685.PCA9685(4, 0x45);
                    ^
Error: Illegal arguments for construction of _exports_PCA9685
```
from early similar issues, this seemed to be a permissions issue but i am running as root

nigeltiany, 2017-09-28:
I think i my case the device is just not available
```shell
-- Logs begin at Thu 2016-02-11 16:28:00 UTC, end at Thu 2017-09-28 18:12:13 UTC. --
Sep 28 18:02:18 odroid libmraa[23238]: i2c5: write_byte_data: Access error: No such device or address
Sep 28 18:02:18 odroid sudo[23234]: pam_unix(sudo:session): session closed for user root
Sep 28 18:02:37 odroid sudo[23250]:   odroid : TTY=pts/1 ; PWD=/home/odroid/Documents/untitled ; USER=root ; 
Sep 28 18:02:37 odroid sudo[23250]: pam_unix(sudo:session): session opened for user root by odroid(uid=0)
Sep 28 18:02:37 odroid libmraa[23254]: libmraa version v1.8.0 initialised by user 'root' with EUID 0

```

Propanu, 2017-09-28:
Could be bad bus/I2C address too, from your initial post I see you only have 2 I2C buses, thus 4 would be out of range. What does mraa write to the system log when you try to initialize the sensor?

nigeltiany, 2017-09-28:
From the [device docs](http://odroid.com/dokuwiki/doku.php?id=en:xu4_hardware_hsi2c) my second i2c object should map to HS:I2C which should be on ```/dev/i2c-5```. The journalctl log says ```i2c5``` rather than ```i2c-5``` is not available. I have switched the code to use index one on the i2c array 
```javascript 
var pca9685Driver = new pca9685.PCA9685(1, 0x45);
```

Propanu, 2017-09-28:
That's just the format of the logged message should not affect functionality, mraa actually tries to open `/dev/i2c-%d` where `%d` is the bus number. Any chance you can run the `i2cdetect` tool on that system to see if the pca9685 shows up on the bus?

nigeltiany, 2017-09-28:
Yes, i must be mulfunctioning

```sudo i2cdetect -r 5
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-5 using read byte commands.
I will probe address range 0x03-0x77.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --            
```

Propanu, 2017-09-28:
Have you had that pca9685 working before? Off the top of my head, possible issues: missing i2c pull-up resistors or floating address pins on the pca9685.

alext-mkrs, 2018-01-28:
This one looks solved to me (or at least no longer relevant) - feel free to comment if further help is needed.

closed 2018-01-28 by alext-mkrs