eclipse-mraa/mraa #828: Add support for MIPS boards based on Mediatek MT7688 chip

sergev, 2017-09-28:
I'm going to add support for MIPS platform and two boards, based on Mediatek MT7688 chip:
(1) Linkit Smart 7688 and Duo (https://labs.mediatek.com/en/platform/linkit-smart-7688)
(2) Omega2 and Omega2+ (https://onion.io/omega2/)

A patch for old version of mraa is available on OpenWRT site (https://github.com/openwrt/packages/blob/master/libs/libmraa/patches/0002-add-mips-support.patch). My plan is to adapt this patch for the current MRAA version, add Omega2 support, and then submit a pull request.

Propanu, 2017-09-28:
Hi @sergev, we were under the impression that the Mediatek 7688 already supports mraa and upm, is that not the case? See the article [here](https://docs.labs.mediatek.com/resource/linkit-smart-7688/en/tutorials/peripheral-support-on-linkit-smart-7688-development-board) for reference.

sergev, 2017-09-28:
Mediatek uses a pretty outdated version mraa-0.8.0.tar.gz with a set of patches applied on top of it (https://github.com/openwrt/packages/tree/master/libs/libmraa). It works fine for Linkit Smart 7688 board. I want to add a support for Omega2 board, so I decided it makes sense to update to the latest MRAA version first. It would be also great to have these sources in the upstream as well.

arfoll, 2017-09-28:
> packages/tree/master/libs/libmraa). It works fine for Linkit Smart 7688

Cool, looking forward to the patches. Let us know if you need any help!

sergev, 2017-09-29:
Here it is: see pull request #829

alext-mkrs, 2017-09-30:
Thanks for the PR, @sergev, I've just went over it and added some comments.
And as the PR is in, let me close this one, we can have a discussion there.

closed 2017-09-30 by alext-mkrs