eclipse-mraa/mraa #837: Problems with building mraa

priestnot, 2017-10-18:
Helo I am trying to build mraa on a upboard.

After problems with swig versions I finaly made it trough cmake. But now make is giving me trouble.
I have made this process with the latest commit on master and the latest tagged version.

Here is theoutput of the make command:
[log.txt](https://github.com/intel-iot-devkit/mraa/files/1395675/log.txt)

arfoll, 2017-10-18:
So your problem may be actually quite simple, your clock is massively off see:
```
make[2]: Warning: File `../src/python/python3/mraa3.i' has modification time 1,5e+04 s in the future
```
Although this is still super odd, the v8 stuff is breaking all over the place but it definately works since I built it with the same node version.... Let's see if with the right clock it'll work.

mgiaco, 2017-11-30:
I know that. So most of the time i get that if something isn´t correct with the time on the board. So the best thing is to checkout a fresh copy and try again. That solved my problem.

alext-mkrs, 2018-01-16:
@priestnot, have you succeeded with building the library after fixing the time offset? Do you need any further help?

mgiaco, 2018-01-17:
Yes sorry I forgot to close

closed 2018-01-17 by arfoll